uhi22/pyPLC #1: rpi python pyPlc.py E S not working

coollhc77, 2023-05-19:
Hello

I'm trying to simulate a pyPlc.
python pyPlc.py P S works, but EVSE does not.
The logs are shown below.

Thanks to you, I was able to try slac using the AR7420.

[image]

uhi22, 2023-05-19:
This may be caused by incomplete implementation of the simulation mode, it tries to use the real ethernet interface, but this seems to go wrong. Could you please test the following two cases: 1. Do you get the same error if a modem is physically connected to the ethernet port and powered? 2. Do you get the same error, if not using the simulation mode, but just sudo python pyPlc.py E?

coollhc77, 2023-05-19:
1. Do you get the same error if a modem is physically connected to the ethernet port and powered?
-> Yes
2. Do you get the same error, if not using the simulation mode, but just sudo python pyPlc.py E?
-> Yes

[image]

[image]

Thank you for your response.

uhi22, 2023-05-19:
In pyPlcTcpSocket.py, could you add after the line 
hostname=socket.gethostname()
a line
print("Your Computer Name is:"+hostname)
to find out the host name?

As temporary workaround, you could kick-out the complete paragraph:
[image]

ArendJanKramer, 2023-05-23:
I ran into the same issue. For me it was that there was no IPV6 address asssigned to the hostname.
Look in /etc/hosts

[image]

Add your hostname in the ::1 definition. For me that solved the issue

uhi22, 2023-06-04:
Next steps: Find out how content of /etc/hosts file under ::1 influences the behavior, and why it works on some devices from scratch. Add exception handling incl clear instruction how to solve.

uhi22, 2023-06-21:
Cannot reproduce the issue on my setup. Let's assume that configuring a host name as described above solves the topic. -> Close.

closed 2023-06-21 by uhi22