uhi22/pyPLC #9: EvseMode: Car closes the connection after AppProtocolResponse

uhi22, 2023-06-27:
Start of the discussion was here: https://github.com/uhi22/pyPLC/issues/6#issuecomment-1601391478

Problem description: After successful SLAC, SDP and protocol handshake, the car closes the TCP connection. This happens stable with the BMW iX3. No problem with Tesla and Hyundai Ioniq.

Root cause: The openV2Gx sets the supportedAppProtocolRes.SchemaID = 1, because we assume, that the car sends a list of supported schemas where the first enty is DIN and has SchemaID 1. The iX3 sends the (one-and-only) DIN schema entry with SchemaID = 0 (which is allowed), and closes the connection after the pyPlc selects the SchemaID 1.

Clean solution would be, that the pyPlc parses the list of announced schemas, selects the entry with DIN, and sends back the SchemaID of that entry back to the car, no matter which value it is.
Potential workaround could be, to send to SchemaID at all. The SchemaID is an optional field, and in best case the vehicle is also happy without it. There is the risk, that some car types will accept this, maybe others not.

uhi22, 2023-06-27:
Pushed the fix of fsmEvse. In best case, this should be the solution for the issue.
To test it, the following steps are necessary:
- pull OpenV2Gx
- compile OpenV2Gx
- pull pyPLC

When running with the iX3, this should look like
`[13641ms] [EVSE] In state WaitForSupportedApplicationProtocolRequest, received (42bytes) = 01 FE 80 01 00 00 00 22 80 00 DB AB 93 71 D3 23 4B 71 D1 B9 81 89 91 89 D1 91 81 89 91 D2 6B 9B 3A 23 2B 30 02 00 00 00 00 40
[13781ms] [EVSE] {
"msgName": "supportedAppProtocolReq",
"info": "34 bytes to convert",
"error": "",
"result": "Vehicle supports 1 protocols. ProtocolEntry#1 ProtocolNamespace=urn:din:70121:2012:MsgDef Version=2.0 SchemaID=0 Priority=1 ",
"schema": "appHandshake",
"AppProtocol_arrayLen": "1",
"NameSpace_0": "urn:din:70121:2012:MsgDef",
"Version_0": "2.0",
"SchemaID_0": "0",
"Priority_0": "1",
"debug": ""
}
[13781ms] [EVSE] The car supports 1 schemas.
[13797ms] [EVSE] The NameSpace urn:din:70121:2012:MsgDef has SchemaID 0
[13797ms] [EVSE] Detected DIN
[13844ms] [EVSE] responding (12bytes) = 01 FE 80 01 00 00 00 04 80 40 00 00
`

and for the Tesla
`[17428ms] [EVSE] In state WaitForSupportedApplicationProtocolRequest, received (74bytes) = 01 FE 80 01 00 00 00 42 80 00 DB AB 93 71 D3 23 4B 71 D1 B9 81 89 91 89 D1 91 81 89 91 D2 6B 9B 3A 23 2B 30 02 00 00 04 04 01 B7 57 26 E3 A7 46 57 36 C6 13 A6 46 96 E3 A3 23 03 13 83 A4 D7 36 74 46 56 60 00 01 C0 10 00 80
[17506ms] [EVSE] {
"msgName": "supportedAppProtocolReq",
"info": "66 bytes to convert",
"error": "",
"result": "Vehicle supports 2 protocols. ProtocolEntry#1 ProtocolNamespace=urn:din:70121:2012:MsgDef Version=2.0 SchemaID=1 Priority=2 ProtocolEntry#2 ProtocolNamespace=urn:tesla:din:2018:MsgDef Version=0.7 SchemaID=2 Priority=1 ",
"schema": "appHandshake",
"AppProtocol_arrayLen": "2",
"NameSpace_0": "urn:din:70121:2012:MsgDef",
"Version_0": "2.0",
"SchemaID_0": "1",
"Priority_0": "2",
"NameSpace_1": "urn:tesla:din:2018:MsgDef",
"Version_1": "0.7",
"SchemaID_1": "2",
"Priority_1": "1",
"debug": ""
}
[17521ms] [EVSE] The car supports 2 schemas.
[17521ms] [EVSE] The NameSpace urn:din:70121:2012:MsgDef has SchemaID 1
[17521ms] [EVSE] The NameSpace urn:tesla:din:2018:MsgDef has SchemaID 2
[17521ms] [EVSE] Detected DIN
[17568ms] [EVSE] responding (12bytes) = 01 FE 80 01 00 00 00 04 80 40 00 40
`

fluppie, 2023-06-27:
Did the steps, but when wanting to start the service (I use these parameters: https://github.com/ArendJanKramer/SmartEVSE-3/blob/modem_docs/manual/Modem.md)

[pyPlc.ini.txt](https://github.com/uhi22/pyPLC/files/11887150/pyPlc.ini.txt)

```
Jun 27 22:46:24 rpievse systemd[1]: Started pyPlc.
Jun 27 22:46:25 rpievse python3.9[8424]: starting in EVSE_MODE
Jun 27 22:46:25 rpievse python3.9[8424]: press Ctrl-C to exit
Jun 27 22:46:25 rpievse python3.9[8424]: initializing pyPlcWorker
Jun 27 22:46:25 rpievse python3.9[8424]: [addressManager] we have local MAC 7A:B7:CC:7C:F9:B9.
Jun 27 22:46:25 rpievse python3.9[8424]: [addressManager] Found 1 link-local IPv6 addresses.
Jun 27 22:46:25 rpievse python3.9[8424]: fe80::956e:4ee7:aeee:9b73
Jun 27 22:46:25 rpievse python3.9[8424]: [addressManager] Local IPv6 is fe80::956e:4ee7:aeee:9b73
Jun 27 22:46:25 rpievse python3.9[8424]: Linux interface is eth1
Jun 27 22:46:25 rpievse python3.9[8424]: [addressManager] will give local MAC 7A:B7:CC:7C:F9:B9
Jun 27 22:46:25 rpievse python3.9[8424]: [addressManager] will give local MAC 7A:B7:CC:7C:F9:B9
Jun 27 22:46:25 rpievse python3.9[8424]: pyPlcIpv6 started with ownMac 7A:B7:CC:7C:F9:B9
Jun 27 22:46:25 rpievse python3.9[8424]: [addressManager] will give local MAC 7A:B7:CC:7C:F9:B9
Jun 27 22:46:25 rpievse python3.9[8424]: udplog started with ownMac 7A:B7:CC:7C:F9:B9
Jun 27 22:46:25 rpievse python3.9[8424]: logging to UDP Syslog is enabled
Jun 27 22:46:25 rpievse python3.9[8424]: sniffer created at eth1
Jun 27 22:46:25 rpievse python3.9[8424]: [67ms] [HARDWAREINTERFACE] Auto detection of serial ports. Available serial ports:
Jun 27 22:46:25 rpievse python3.9[8424]: [82ms] [HARDWAREINTERFACE] ignoring /dev/ttyAMA0, because this is not an USB serial port
Jun 27 22:46:25 rpievse python3.9[8424]: [83ms] [HARDWAREINTERFACE] ERROR: No serial ports found. No hardware interaction possible.
Jun 27 22:46:25 rpievse python3.9[8424]: Traceback (most recent call last):
Jun 27 22:46:25 rpievse python3.9[8424]:   File "/home/pi/myprogs/pyPlc/evseNoGui.py", line 62, in <module>
Jun 27 22:46:25 rpievse python3.9[8424]:     worker=pyPlcWorker.pyPlcWorker(cbAddToTrace, cbShowStatus, myMode, 0, socStatusCallback)
Jun 27 22:46:25 rpievse python3.9[8424]:   File "/home/pi/myprogs/pyPlc/pyPlcWorker.py", line 34, in __init__
Jun 27 22:46:25 rpievse python3.9[8424]:     self.hardwareInterface = hardwareInterface.hardwareInterface(self.workerAddToTrace, self.showStatus)
Jun 27 22:46:25 rpievse python3.9[8424]:   File "/home/pi/myprogs/pyPlc/hardwareInterface.py", line 259, in __init__
Jun 27 22:46:25 rpievse python3.9[8424]:     self.initPorts()
Jun 27 22:46:25 rpievse python3.9[8424]:   File "/home/pi/myprogs/pyPlc/hardwareInterface.py", line 217, in initPorts
Jun 27 22:46:25 rpievse python3.9[8424]:     self.canbus = can.interface.Bus(bustype='socketcan', channel="can0", can_filters = filters)
Jun 27 22:46:25 rpievse python3.9[8424]: AttributeError: module 'can' has no attribute 'interface'
Jun 27 22:46:25 rpievse systemd[1]: pyPlc.service: Main process exited, code=exited, status=1/FAILURE
Jun 27 22:46:25 rpievse systemd[1]: pyPlc.service: Failed with result 'exit-code'.
Jun 27 22:46:25 rpievse systemd[1]: pyPlc.service: Consumed 1.015s CPU time.
Jun 27 22:46:25 rpievse systemd[1]: pyPlc.service: Scheduled restart job, restart counter is at 5.
Jun 27 22:46:25 rpievse systemd[1]: Stopped pyPlc.
Jun 27 22:46:25 rpievse systemd[1]: pyPlc.service: Consumed 1.015s CPU time.
Jun 27 22:46:25 rpievse systemd[1]: pyPlc.service: Start request repeated too quickly.
Jun 27 22:46:25 rpievse systemd[1]: pyPlc.service: Failed with result 'exit-code'.
Jun 27 22:46:25 rpievse systemd[1]: Failed to start pyPlc.
```

Also tried to install can
`sudo python -m pip install --upgrade can `
and
`python -m pip install --upgrade can`

uhi22, 2023-06-28:
This sounds like chademo is enabled in the ini file. Just replace it with "none".

fluppie, 2023-06-28:
Ok cool, looks like we are on the right track :-). Only the energycapacity should be around 74 kWh netto or 80 kWh bruto.
I think ABRP is able to read this via the OBDLink CX dongle.

```
Jun 28 08:28:42 rpievse python3.9[2137]: [257521ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
Jun 28 08:28:45 rpievse python3.9[2137]: [260196ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
Jun 28 08:28:48 rpievse python3.9[2137]: [262873ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
Jun 28 08:28:49 rpievse python3.9[2137]: [264091ms] received SLAC_PARAM.REQ
Jun 28 08:28:49 rpievse python3.9[2137]: [addressManager] pev has MAC EC:65:CC:B6:85:DC
Jun 28 08:28:49 rpievse python3.9[2137]: [264092ms] [EVSE] transmitting CM_SLAC_PARAM.CNF
Jun 28 08:28:49 rpievse python3.9[2137]: [264338ms] received MNBC_SOUND.IND
Jun 28 08:28:49 rpievse python3.9[2137]: [264339ms] received MNBC_SOUND.IND
Jun 28 08:28:49 rpievse python3.9[2137]: [264422ms] received MNBC_SOUND.IND
Jun 28 08:28:49 rpievse python3.9[2137]: [264423ms] received MNBC_SOUND.IND
Jun 28 08:28:49 rpievse python3.9[2137]: [264424ms] received MNBC_SOUND.IND
Jun 28 08:28:49 rpievse python3.9[2137]: [264506ms] received MNBC_SOUND.IND
Jun 28 08:28:49 rpievse python3.9[2137]: [264507ms] received MNBC_SOUND.IND
Jun 28 08:28:49 rpievse python3.9[2137]: [264508ms] received MNBC_SOUND.IND
Jun 28 08:28:50 rpievse python3.9[2137]: [264590ms] received MNBC_SOUND.IND
Jun 28 08:28:50 rpievse python3.9[2137]: [264591ms] received MNBC_SOUND.IND
Jun 28 08:28:50 rpievse python3.9[2137]: [264592ms] [EVSE] transmitting ATTEN_CHAR.IND
Jun 28 08:28:50 rpievse python3.9[2137]: [265486ms] received SLAC_MATCH.REQ
Jun 28 08:28:50 rpievse python3.9[2137]: [265487ms] [EVSE] transmitting SLAC_MATCH.CNF
Jun 28 08:28:51 rpievse python3.9[2137]: [265569ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
Jun 28 08:28:53 rpievse python3.9[2137]: [268247ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
Jun 28 08:28:55 rpievse python3.9[2137]: [addressManager] pev has IP fe80:0000:0000:0000:ee65:ccff:feb6:85dc
Jun 28 08:28:55 rpievse python3.9[2137]: V2GTP (10bytes) = 01 FE 90 00 00 00 00 02 10 00
Jun 28 08:28:55 rpievse python3.9[2137]: Ok, this was a valid SDP request. We are the SECC. Sending SDP response.
Jun 28 08:28:55 rpievse python3.9[2137]: SDP payload (20bytes) = FE 80 00 00 00 00 00 00 DB 7C 36 04 B2 2B C7 CF 3B 0E 10 00
Jun 28 08:28:55 rpievse python3.9[2137]: V2Gframe (28bytes) = 01 FE 90 01 00 00 00 14 FE 80 00 00 00 00 00 00 DB 7C 36 04 B2 2B C7 CF                                                                                                         3B 0E 10 00
Jun 28 08:28:55 rpievse python3.9[2137]: [269877ms] Connection from ('fe80::ee65:ccff:feb6:85dc', 49153, 0, 4)
Jun 28 08:28:55 rpievse python3.9[2137]: [SNIFFER] EXI from 49153 to 15118 (34bytes) = 80 00 DB AB 93 71 D3 23 4B 71 D1 B9 81 89 91 8                                                                                                        9 D1 91 81 89 91 D2 6B 9B 3A 23 2B 30 02 00 00 00 00 40
Jun 28 08:28:55 rpievse python3.9[2137]: [269911ms] [EVSE] In state WaitForSupportedApplicationProtocolRequest, received (42bytes) =                                                                                                         01 FE 80 01 00 00 00 22 80 00 DB AB 93 71 D3 23 4B 71 D1 B9 81 89 91 89 D1 91 81 89 91 D2 6B 9B 3A 23 2B 30 02 00 00 00 00 40
Jun 28 08:28:55 rpievse python3.9[2137]: [269941ms] [EVSE] {
Jun 28 08:28:55 rpievse python3.9[2137]: "msgName": "supportedAppProtocolReq",
Jun 28 08:28:55 rpievse python3.9[2137]: "info": "34 bytes to convert",
Jun 28 08:28:55 rpievse python3.9[2137]: "error": "",
Jun 28 08:28:55 rpievse python3.9[2137]: "result": "Vehicle supports 1 protocols. ProtocolEntry#1 ProtocolNamespace=urn:din:70121:201                                                                                                        2:MsgDef Version=2.0 SchemaID=0 Priority=1 ",
Jun 28 08:28:55 rpievse python3.9[2137]: "schema": "appHandshake",
Jun 28 08:28:55 rpievse python3.9[2137]: "AppProtocol_arrayLen": "1",
Jun 28 08:28:55 rpievse python3.9[2137]: "NameSpace_0": "urn:din:70121:2012:MsgDef",
Jun 28 08:28:55 rpievse python3.9[2137]: "Version_0": "2.0",
Jun 28 08:28:55 rpievse python3.9[2137]: "SchemaID_0": "0",
Jun 28 08:28:55 rpievse python3.9[2137]: "Priority_0": "1",
Jun 28 08:28:55 rpievse python3.9[2137]: "debug": ""
Jun 28 08:28:55 rpievse python3.9[2137]: }
Jun 28 08:28:55 rpievse python3.9[2137]: [269944ms] [EVSE] The car supports 1 schemas.
Jun 28 08:28:55 rpievse python3.9[2137]: [269945ms] [EVSE] The NameSpace urn:din:70121:2012:MsgDef has SchemaID 0
Jun 28 08:28:55 rpievse python3.9[2137]: [269946ms] [EVSE] Detected DIN
Jun 28 08:28:55 rpievse python3.9[2137]: [269960ms] [EVSE] responding (12bytes) = 01 FE 80 01 00 00 00 04 80 40 00 00
Jun 28 08:28:55 rpievse python3.9[2137]: [269962ms] [EVSE] from 0 entering 1
Jun 28 08:28:55 rpievse python3.9[2137]: [SNIFFER] EXI from 15118 to 49153 (4bytes) = 80 40 00 00
Jun 28 08:28:55 rpievse python3.9[2137]: [270021ms] [EVSE] In state WaitForSessionSetupRequest, received (22bytes) = 01 FE 80 01 00 0                                                                                                        0 00 0E 80 9A 00 40 11 D0 1B B1 97 32 DA 17 70 00
Jun 28 08:28:55 rpievse python3.9[2137]: [270041ms] [EVSE] {
Jun 28 08:28:55 rpievse python3.9[2137]: "msgName": "SessionSetupReq",
Jun 28 08:28:55 rpievse python3.9[2137]: "info": "14 bytes to convert",
Jun 28 08:28:55 rpievse python3.9[2137]: "error": "",
[... cut ...]
Jun 28 08:41:10 rpievse python3.9[2137]: [1004679ms] The socket is linked the following IP addresses:
Jun 28 08:41:10 rpievse python3.9[2137]: [1004680ms] ::1
Jun 28 08:41:10 rpievse python3.9[2137]: [1004680ms] ::1
Jun 28 08:41:10 rpievse python3.9[2137]: [1004680ms] ::1
Jun 28 08:41:10 rpievse python3.9[2137]: [1004792ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
Jun 28 08:41:12 rpievse python3.9[2137]: [1007471ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
Jun 28 08:41:15 rpievse python3.9[2137]: [1010154ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5

```

uhi22, 2023-06-28:
Great, so I close this issue.
Regarding the EnergyCapacity: The car does not provide it, it says "EVEnergyCapacity_isUsed": "0". The remaining inconsistency of the pyPLC is, that it reports a 27kWh default value, which could be improved. The value is configured in pyPlc.ini, with soc_fallback_energy_capacity = 2700. Needs to be decided, whether this configurable value is good enough or a better strategy is necessary. May worth a new issue.

closed 2023-06-28 by uhi22