uhi22/pyPLC #12: [EVSE] Weird behaviour during communication when car is 100% charged

arpiecodes, 2023-07-13:
Hello!

I'd like to share a weird edge case I encountered today. The car was charged 100% at this point, but for some reason doesn't want to communicate with pyPLC anymore. I'm puzzled as everything seems to be OK.

````
Jul 14 00:08:31 pievse python3.9[21553]: [469700ms] Connection from ('fe80::xxx:xxx:xxx:xxx', 49160, 0, 3)
Jul 14 00:08:31 pievse python3.9[21553]: [469751ms] [EVSE] In state WaitForSupportedApplicationProtocolRequest, received (76bytes) = 01 FE 80 01 00 00 00 44 80 00 DB AB 93 71 D3 23 4B 71 D1 B9 81 89 91 89 D1 91 81 89 91 D2 6B 9B 3A 23 2B 30 02 00 00 04 00 01 D7 57 26 E3 A6 97 36 F3 A3 13 53 13 13 83 A3 23 A3 23 03 13 33 A4 D7 36 74 46 56 60 04 00 00 00 08 80
Jul 14 00:08:31 pievse python3.9[21553]: [469813ms] [EVSE] {
Jul 14 00:08:31 pievse python3.9[21553]: "msgName": "supportedAppProtocolReq",
Jul 14 00:08:31 pievse python3.9[21553]: "info": "68 bytes to convert",
Jul 14 00:08:31 pievse python3.9[21553]: "error": "",
Jul 14 00:08:31 pievse python3.9[21553]: "result": "Vehicle supports 2 protocols. ProtocolEntry#1 ProtocolNamespace=urn:din:70121:2012:MsgDef Version=2.0 SchemaID=1 Priority=1 ProtocolEntry#2 ProtocolNamespace=urn:iso:15118:2:2013:MsgDef Version=2.0 SchemaID=0 Priority=2 ",
Jul 14 00:08:31 pievse python3.9[21553]: "schema": "appHandshake",
Jul 14 00:08:31 pievse python3.9[21553]: "debug": ""
Jul 14 00:08:31 pievse python3.9[21553]: }
Jul 14 00:08:31 pievse python3.9[21553]: [469830ms] [EVSE] ERROR: Could not decode the supportedAppProtocolReq
Jul 14 00:08:31 pievse python3.9[21553]: [469834ms] [EVSE] Error: The connected car does not support DIN. At the moment, the pyPLC only supports DIN.
Jul 14 00:08:33 pievse python3.9[21553]: [471689ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
Jul 14 00:08:33 pievse python3.9[21553]: [471698ms] connection closed
Jul 14 00:08:33 pievse python3.9[21553]: [471701ms] [EVSE] re-initializing fsmEvse due to broken connection
````

Also; for some reason my car likes to return EVRESSSOC = 0 along with EnergyRequest = 0 if it's fully charged;

````
l 13 23:46:58 pievse python3.9[14583]: [1293353855ms] [EVSE] In state WaitForFlexibleRequest, received (46bytes) = 01 FE 80 01 00 00 00 26 80 9A 02 00 >
Jul 13 23:46:58 pievse python3.9[14583]: [1293353929ms] [EVSE] {
Jul 13 23:46:58 pievse python3.9[14583]: "msgName": "ChargeParameterDiscoveryReq",
Jul 13 23:46:58 pievse python3.9[14583]: "info": "38 bytes to convert",
Jul 13 23:46:58 pievse python3.9[14583]: "error": "",
Jul 13 23:46:58 pievse python3.9[14583]: "result": "",
Jul 13 23:46:58 pievse python3.9[14583]: "schema": "DIN",
Jul 13 23:46:58 pievse python3.9[14583]: "g_errn": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "header.SessionID": "0102030405060708",
Jul 13 23:46:58 pievse python3.9[14583]: "header.Notification_isUsed": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "header.Signature_isUsed": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVRequestedEnergyTransferType": "3",
Jul 13 23:46:58 pievse python3.9[14583]: "EVChargeParameter_isUsed": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "DC_EVChargeParameter_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "DC_EVStatus.EVRESSSOC": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "DC_EVStatus.EVReady": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVCabinConditioning_isUsed": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVRESSConditioning_isUsed": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVErrorCode": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "DC_EVErrorCodeText": "NO_ERROR",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumCurrentLimit.Value": "3750",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumCurrentLimit.Multiplier": "-1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumCurrentLimit.Unit_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumCurrentLimit.Unit": "3",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumPowerLimit_isUsed": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumPowerLimit.Value": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumPowerLimit.Multiplier": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumPowerLimit.Unit_isUsed": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumPowerLimit.Unit": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumVoltageLimit.Value": "470",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumVoltageLimit.Multiplier": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumVoltageLimit.Unit_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVMaximumVoltageLimit.Unit": "5",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyCapacity_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyCapacity.Value": "6890",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyCapacity.Multiplier": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyCapacity.Unit_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyCapacity.Unit": "9",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyRequest_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyRequest.Value": "0",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyRequest.Multiplier": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyRequest.Unit_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "EVEnergyRequest.Unit": "9",
Jul 13 23:46:58 pievse python3.9[14583]: "FullSOC_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "FullSOC": "100",
Jul 13 23:46:58 pievse python3.9[14583]: "BulkSOC_isUsed": "1",
Jul 13 23:46:58 pievse python3.9[14583]: "BulkSOC": "80",
Jul 13 23:46:58 pievse python3.9[14583]: "debug": "Line9057Line9260Line400"
Jul 13 23:46:58 pievse python3.9[14583]: }
Jul 13 23:46:58 pievse python3.9[14583]: [1293353959ms] [EVSE] Received ChargeParameterDiscoveryReq. Extracting SoC parameters via DC
Jul 13 23:46:58 pievse python3.9[14583]: Received SoC status from ChargeParameterDiscoveryReq.
Jul 13 23:46:58 pievse python3.9[14583]:     Current SoC 0%
Jul 13 23:46:58 pievse python3.9[14583]:     Full SoC 100%
Jul 13 23:46:58 pievse python3.9[14583]:     Energy capacity 68900 Wh
Jul 13 23:46:58 pievse python3.9[14583]:     Energy request -1 Wh
Jul 13 23:46:58 pievse python3.9[14583]:     EVCCID xxxxxxxxxx
````

I could create a workaround for this in the code to assume `current_soc` = `full_soc` when both values are explicitly 0. Let me know if that's acceptable.

uhi22, 2023-07-14:
So we have two topics here:

1. The pyPlc complains "Could not decode the supportedAppProtocolReq". This is related to the change   https://github.com/uhi22/pyPLC/issues/9#issuecomment-1610098576, where the evaluation of the SchemaIDs was improved. It needs to have a consistent (e.g. the latest versions) of pyPLC and openV2Gx. In the log, we see that the decoder does not provide the field "NameSpace_0", so it is the old version. Just pull the latest openV2Gx and compile it.

2. Yes, seems it is needed to add a workaround, to interpret the 0% SOC as 100% SOC. But there is the risk, that a real 0% is wrongly "corrected" to 100%, so we need a reliable criteria to distinguish the two cases. I'd propose to add a kind of "model detection", e.g. based on the EVEnergyCapacity and EVMaximumVoltageLimit. Something like
if EVEnergyCapacity==68900 and EVMaximumVoltageLimit==470):
# special handling for <manufacturer> <model>
if (EVRESSSOC==0 and EVEnergyRequest_isUsed == 1 and EVEnergyRequest == 0):
# The car is not requesting energy, this happens if the battery is full. Correct the SOC to "Full":
current_soc = full_soc
Maybe this approach is suitable / necessary for multiple manufacturers or car models, so the check for the specific model may not be necessary. But as long as we do not know, I would go the safe way.
Feel free to make a proposal pull request,

uhi22, 2024-05-31:
Closing due to inactivity.

closed 2024-05-31 by uhi22