uhi22/pyPLC #13: Question for very start

Electricalfan, 2023-08-17:
Hi, where can I find list of standard PLC messages? I understand EV and EVSE send them via TCP on CP line and I can read them, but I'd like to know whole list of these messages and how to compose and send them. Where could I find this information?

uhi22, 2023-08-21:
There are several layers involved, and all all have its own specification.
At top level, there is the ISO15118 or DIN 70121. Each of these specifiy a certain set of messages which are exchanged between car and charger. Both norms are similar, but not identical. The handshake process decides, which of both is used, depending on the capabilities of the car and of the charger. These two norms are "open" but not for free. There are different ways to purchase them, e.g. https://www.beuth.de/de/technische-regel/din-spec-70121/224350045 . The message definition can be seen in the header files in https://github.com/Martin-P/OpenV2G.

The level below is the Vehicle-To-Grid-Transport-Protocol. This is simply adding a header with a fix pattern and a length information. I think this is also specified in the above norms.

The level below TCP, then IPv6, and ethernet. Further below the PLC, for which I do not know the spec, but this is not really relevant.

It makes sense to read through the references in https://github.com/uhi22/pyPLC#references and https://www.goingelectric.de/wiki/CCS-Technische-Details/

One starting point can be https://ltu.diva-portal.org/smash/get/diva2:1546405/FULLTEXT01.pdf which explains the matching process on PLC level.

uhi22, 2023-09-06:
Since there is no feedback, assuming the questions are answered. Closing.

closed 2023-09-06 by uhi22