uhi22/pyPLC #14: Close connection due to broken connection

MarkG-PE, 2023-09-10:
@uhi22 
Brought this issue over from https://github.com/SmartEVSE/SmartEVSE-3/issues/25#issuecomment-1666231234.
I tested the latest update with change to OpenV2gx on BMW Ix3. 
The connection performs the same as previous with a shutdown after charge parameter discovery.
Wireshark log attached.
[120ohmPP.pcapng.txt](https://github.com/uhi22/pyPLC/files/12568541/120ohmPP.pcapng.txt)
I note that the SOC was 80%, but given the previous 45% SOC performed the same, i think this issue is not related to the EV's SOC value.

MarkG-PE, 2023-09-10:
@uhi22, I think this might be due to me using direct electrical connections (alligator clips) to the car and not a plug.
So the car thinks it cant lock the plug properly.
I will get a plug and test it through the plug rather than using direct connections.

uhi22, 2023-09-11:
You could use your standard type2 charging cable to trick the car.

MarkG-PE, 2023-09-11:
I will test the type2  plug with the extension cable that come with the car.
I would like to however interpret the last response sent from the EVSE to the car, which if i decode via V2G Viewer Pro, then it gives: 
DC_EVSEChargeParameter.DC_EVSEStatus.EVSEIsolationStatus|Invalid
DC_EVSEChargeParameter.EVSEMaximumCurrentLimit|200 A
DC_EVSEChargeParameter.EVSEMaximumPowerLimit|10000 W
DC_EVSEChargeParameter.EVSEMaximumVoltageLimit|450 V
DC_EVSEChargeParameter.EVSEMinimumVoltageLimit|200 V
DC_EVSEChargeParameter.EVSECurrentRegulationTolerance|5 A
DC_EVSEChargeParameter.EVSEPeakCurrentRipple|5 A

Can you tell me where these values are encoded to respond to the EV, and can i change them to more closely align with the EV response.?

Also, can i make the isolation status valid at this step of charge param discovery.?

uhi22, 2023-09-11:
These values are hardcoded in the OpenV2Gx, in function encodeChargeParameterDiscoveryResponse(). You can adapt the values there, and re-compile the openV2Gx. If the values are needed to be changed dynamically, they could given via command line parameter from the pyPLC to the OpenV2Gx, similar like done for the EVRESSSOC in encodeCableCheckRequest().
But: I see no reason why the car should reject the charging with these parameters. Ok, maybe 10kW maximum power is not plausible for a fast charger.
The isolation status is intentionally invalid at the chargeParameterDiscovery. This is what I saw at public chargers, and it makes sense. The isolation status will be checked later in the sequence, during the CableCheck. Then the charger needs to report a valid isolation status to continue.

uhi22, 2023-09-11:
In the pcap above, the car actively closes the connection after the chargeParameterDiscoveryResponse, ~30ms after it acknowledge the response. This means, it is not a broken connection, it is intentionally closed connection.

Normally, according to https://github.com/uhi22/pyPLC#example-flow after the ChargeParameterDiscoveryResponse (checkpoint 545), the car would change to StateC and lock the connector. I think it is unlikely, that it comes within 30ms to the conclusion, that the connector lock failed. So more likely is, that something with the voltage/resistors on the CP line is wrong, and that the car is unhappy with this. Or the PP line. I use 1.5kOhms, which I also measured on public chargers.

MarkG-PE, 2023-09-29:
its not the plug unfortunately .
I tried with the plug and same result. 
So i tried connecting the modem across a real connection with the car and running wireshark.
saw a lot of 'encrypted messages'. Then read the readme file on pyPLC and noted i think i should be running pyPLC in Listening Mode while running wireshark and having my modem connected over the evse and car CP-PE lines. Is this correct?? If so i will go out and give it another run.

MarkG-PE, 2023-09-29:
note also, i am also using 1.5kohms on the PP to Earth.
So my issue is not that.

uhi22, 2023-09-30:
Cannot promise that the listen-mode really helps. It was intended for exactly this purpose, but not maintained since a long time. The goal is to extract the key from the slac sequence, set the key in sniffer modem, so it can decode the traffic. Would be interesting what "encrypted traffic" you see, and using which modem type. Maybe this brings the sniffing topic further...

MarkG-PE, 2023-10-11:
I didn't have the DC Connections shorted through from the plug to the car, just the CP, PP and Earth so naturally at some point this was going to fail. It was a pretty messy session with regard to loose connections and the like, but i believe the major part of Listen Mode is demonstrated with regard to getting through SLAC at line 138 and then seeing Encrypted Payload at lines 142, 144, 145, 147 & 148. 
I will make the plug connections a bit better and also make through the DC cables, but that might be a few days away. So as not to keep the information on hand without sharing it, i will provide what i have now.
[Ix3_coms_only_Listen.pcapng.txt](https://github.com/uhi22/pyPLC/files/12869611/Ix3_coms_only_Listen.pcapng.txt)

uhi22, 2023-10-12:
We see parts of the SLAC sequence, this is good and bad. Ideally, we would see all the SLAC messages from both sides, but mostly we see the car (with the Panasonic MAC address). The last message of it is the SLAC_MATCH.REQ, so it wants to pair with the charger (with the DeltaEle MAC). The charger does most likely confirm with a SLAC_MATCH.CNF (Checkpoint 155 in https://github.com/uhi22/pyPLC#example-flow) . But we do not see this in the trace, most likely it is just filtered-out by the sniffer modem, because it is no broadcast, but an unicast from the DeltaEle to the Panasonic. The pitty is, that the SLAC_MATCH.CNF contains the network key (NMK), that we would need to setup in the sniffer modem, to join the AVLN. The idea is: The pyPLC takes the NMK from the SLAC_MATCH.CNF, and sends a Set_Key with this NMK to the modem. But this does not work, if there is no SLAC_MATCH.CNF received.
Conclusion: To be successfull with the ListenMode, we need to search for a modem which routes the traffic to its ethernet port.
The modem which I used here: https://www.goingelectric.de/wiki/CCS-Technische-Details/ is able to at least see the SLAC traffic from both sides. But this still does not help, because it does not route the unicast IP traffic.

MarkG-PE, 2023-10-12:
Do you think it is possible, to make the pyPLC modem have the same mac address as the car and listen in on the conversation that way. Then as the EVSE sends to NMK, the pyPLC can listen to it and decode the rest of the conversation from the EVSE. It might miss some messages from the PEV to the EVSE, but it would be able to receive all the messages from the EVSE and then the pyPLC could be made to answer the PEV similar to the EVSE.

MarkG-PE, 2023-10-24:
Noting the IX3 uses DIN standard and i think it doesn't support SASchedules, can you advise an understanding of the ChargeParameterDiscoveryRes response with regard to SASchedules.
I am wondering if this response is valid : 
SASchedules.noContent                                                                                                                                              | 16626852

or the response should be fed back as: 

SAScheduleList.SAScheduleTuple.array[0].SAScheduleTupleID                                                                                  | 0
SAScheduleList.SAScheduleTuple.array[0].PMaxSchedule.PMaxScheduleID                                                              | 0
SAScheduleList.SAScheduleTuple.array[0].PMaxSchedule.PMaxScheduleEntry.array[0].RelativeTimeInterval.start    | 0
SAScheduleList.SAScheduleTuple.array[0].PMaxSchedule.PMaxScheduleEntry.array[0].PMax                                    | 0
SAScheduleList.SAScheduleTuple                                                                                                                                | 0

I noted some reported pcap files from other cars gave the more detailed list with zeros, but i haven't found a wireshark trace for a DIN standard car giving this parameter in a successful run.

Mustafa19981903, 2024-01-15:
@uhi22 @MarkG-PE My car is ix3 and I got the same error. Did you find a solution? 
The tool output is as follows. Also, when I use the software as a simulation, DC_EVSEStatus.EVSEIselationStatus=0 is set to 0, I want to change it to 1, and I also send the EVSE maximumpowerlimit value as 10. How can I change these values in the pyPlc software?

"msgName": "ChargeParameterDiscoveryReq",
"info": "36 bytes to convert", 
"error": "",
"result": "",
"schema": "DIN",
"g_errn": "0",
"header.SessionID": "0102030405060708",
"header.Notification_isUsed": "0",
"header.Signature_isUsed": "0",
"EVRequestedEnergyTransferType": "3",
"EVChargeParameter_isUsed": "0",
"DC_EVChargeParameter_isUsed": "1",
"DC_EVStatus.EVRESSSOC": "71",
"DC_EVStatus.EVReady": "0",
"EVCabinConditioning_isUsed": "1",
"EVRESSConditioning_isUsed": "1",
"EVErrorCode": "0",
"DC_EVErrorCodeText": "NO_ERROR",
"EVMaximumCurrentLimit.Value": "5000",
"EVMaximumCurrentLimit.Multiplier": "-1",
"EVMaximumCurrentLimit.Unit_isUsed": "1",
"EVMaximumCurrentLimit.Unit": "3",
"EVMaximumPowerLimit_isUsed": "1",
"EVMaximumPowerLimit.Value": "25000",
"EVMaximumPowerLimit.Multiplier": "1",
"EVMaximumPowerLimit.Unit_isUsed": "1",
"EVMaximumPowerLimit.Unit": "7",
"EVMaximumVoltageLimit.Value": "5000",
"EVMaximumVoltageLimit.Multiplier": "-1",
"EVMaximumVoltageLimit.Unit_isUsed": "1",
"EVMaximumVoltageLimit.Unit": "5",
"EVEnergyCapacity_isUsed": "0",
"EVEnergyCapacity.Value": "0",
"EVEnergyCapacity.Multiplier": "0",
"EVEnergyCapacity.Unit_isUsed": "0",
"EVEnergyCapacity.Unit": "0",
"EVEnergyRequest_isUsed": "0",
"EVEnergyRequest.Value": "0",
"EVEnergyRequest.Multiplier": "0",
"EVEnergyRequest.Unit_isUsed": "0",
"EVEnergyRequest.Unit": "0",
"FullSOC_isUsed": "1",
"FullSOC": "80",
"BulkSOC_isUsed": "1",
"BulkSOC": "80",
"debug": "Line9057Line9090Line9169Line9226Line9260Line400"
}
[127458ms] [EVSE] Received ChargeParameterDiscoveryReq. Extracting SoC parameters via DC
[127478ms] [EVSE] responding (55bytes) = 01 FE 80 01 00 00 00 2F 80 9A 02 00 40 80 C1 01 41 81 C2 10 80 00 48 00 40 00 00 C0 C3 20 04 0C 0E 01 40 60 A1 84 06 06 06 00 20 60 A1 90 02 03 03 00 50 30 30 05 10 
[127481ms] [EVSE] from 4 entering 4
[SNIFFER] EXI from 49154 to 15118 (36bytes) = 80 9A 02 00 40 80 C1 01 41 81 C2 10 71 90 00 00 08 E0 40 61 10 4E 04 07 0A 8C 30 10 20 50 88 27 12 80 50 00 
[SNIFFER] EXI from 15118 to 49154 (47bytes) = 80 9A 02 00 40 80 C1 01 41 81 C2 10 80 00 48 00 40 00 00 C0 C3 20 04 0C 0E 01 40 60 A1 84 06 06 06 00 20 60 A1 90 02 03 03 00 50 30 30 05 10 
[127519ms] connection closed
[127538ms] [EVSE] re-initializing fsmEvse due to broken connection
[127539ms] [EVSE] re-initializing fsmEvse
[127539ms] Trying to reset the TCP socket
[127548ms] pyPlcTcpSocket listening on port 15118
[129972ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
[132758ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
[135501ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5

MarkG-PE, 2024-01-15:
I have tried a lot of variations in parameter settings to match a more typical charger, but no luck yet.
I am stuck on this at the moment.

Mustafa19981903, 2024-01-16:
@MarkG-PE I  @uhi22  changed the maximum power value. Maybe the vehicle does not like the maximum power value and terminates the charge. But there is a situation like this: Zoe vehicle has reached the precharge stage. I wonder why this is happening?

uhi22, 2024-01-16:
You wonder why what is happening? That the Zoe reaches the precharge? I guess each car manufacturer has some own understanding which plausibilizations are necessary, and which reaction is necessary. And obviously, the Zoe is happy with the data provided by pyPLC, and also others, which are documented here: https://openinverter.org/forum/viewtopic.php?t=3551 (Ioniq, MG4, Tesla, BYD atto). For the Everest project there are also "mixed" results, documented here: https://github.com/EVerest/logfiles
To make a step further, I would propose to summarize the obervation again, somehow I'm lost with the multiple mixed topics.
- Which car we are talking about?
- Until which step (message name) it works, and which step fails?
- TCP dump of the situation
- What has been already tried to improve?
Maybe the log files from https://github.com/EVerest/logfiles can be used for comparision. Not sure whether they cover the same vehicle type / version.

Mustafa19981903, 2024-01-16:
@uhi22 @MarkG-PE The first message part is zoe, the second message part is bmw ix3 zoe precharge, but it stays in the bmwix3 chargeparameterdiscovery part and the charging ends.

I want to switch to the precharge section with a BMW ix3. I had the opportunity to try Togg and I stuck to the chargeparameter section with Togg.

[25299ms] [EVSE] {
"msgName": "PreChargeReq",
"info": "24 bytes to convert", 
"error": "",
"result": "",
"schema": "DIN",
"g_errn": "0",
"header.SessionID": "0102030405060708",
"header.Notification_isUsed": "0",
"header.Signature_isUsed": "0",
"DC_EVStatus.EVReady": "1",
"DC_EVStatus.EVErrorCode": "0",
"DC_EVErrorCodeText": "NO_ERROR",
"DC_EVStatus.EVRESSSOC": "76",
"EVTargetVoltage.Multiplier": "-1",
"EVTargetVoltage.Value": "3744",
"EVTargetVoltage.Unit": "V",
"EVTargetCurrent.Multiplier": "-1",
"EVTargetCurrent.Value": "10",
"EVTargetCurrent.Unit": "A",
"debug": "Line9057Line9260Line484"
}
[25305ms] [EVSE] EV wants EVTargetVoltage 374.40000000000003
[25325ms] [EVSE] responding (30bytes) = 01 FE 80 01 00 00 00 16 80 9A 02 00 40 80 C1 01 41 81 C2 11 60 00 40 00 00 01 93 E4 08 00 
[25331ms] [EVSE] from 4 entering 4
[SNIFFER] EXI from 50442 to 15118 (24bytes) = 80 9A 02 00 40 80 C1 01 41 81 C2 11 51 40 13 00 81 42 80 74 08 0C 02 80 
[SNIFFER] EXI from 15118 to 50442 (22bytes) = 80 9A 02 00 40 80 C1 01 41 81 C2 11 60 00 40 00 00 01 93 E4 08 00 
[25406ms] [EVSE] In state WaitForFlexibleRequest, received (32bytes) = 01 FE 80 01 00 00 00 18 80 9A 02 00 40 80 C1 01 41 81 C2 11 51 40 13 00 81 42 80 74 08 0C 02 80

[127457ms] [EVSE] {
"msgName": "ChargeParameterDiscoveryReq",
"info": "36 bytes to convert", 
"error": "",
"result": "",
"schema": "DIN",
"g_errn": "0",
"header.SessionID": "0102030405060708",
"header.Notification_isUsed": "0",
"header.Signature_isUsed": "0",
"EVRequestedEnergyTransferType": "3",
"EVChargeParameter_isUsed": "0",
"DC_EVChargeParameter_isUsed": "1",
"DC_EVStatus.EVRESSSOC": "71",
"DC_EVStatus.EVReady": "0",
"EVCabinConditioning_isUsed": "1",
"EVRESSConditioning_isUsed": "1",
"EVErrorCode": "0",
"DC_EVErrorCodeText": "NO_ERROR",
"EVMaximumCurrentLimit.Value": "5000",
"EVMaximumCurrentLimit.Multiplier": "-1",
"EVMaximumCurrentLimit.Unit_isUsed": "1",
"EVMaximumCurrentLimit.Unit": "3",
"EVMaximumPowerLimit_isUsed": "1",
"EVMaximumPowerLimit.Value": "25000",
"EVMaximumPowerLimit.Multiplier": "1",
"EVMaximumPowerLimit.Unit_isUsed": "1",
"EVMaximumPowerLimit.Unit": "7",
"EVMaximumVoltageLimit.Value": "5000",
"EVMaximumVoltageLimit.Multiplier": "-1",
"EVMaximumVoltageLimit.Unit_isUsed": "1",
"EVMaximumVoltageLimit.Unit": "5",
"EVEnergyCapacity_isUsed": "0",
"EVEnergyCapacity.Value": "0",
"EVEnergyCapacity.Multiplier": "0",
"EVEnergyCapacity.Unit_isUsed": "0",
"EVEnergyCapacity.Unit": "0",
"EVEnergyRequest_isUsed": "0",
"EVEnergyRequest.Value": "0",
"EVEnergyRequest.Multiplier": "0",
"EVEnergyRequest.Unit_isUsed": "0",
"EVEnergyRequest.Unit": "0",
"FullSOC_isUsed": "1",
"FullSOC": "80",
"BulkSOC_isUsed": "1",
"BulkSOC": "80",
"debug": "Line9057Line9090Line9169Line9226Line9260Line400"
}
[127458ms] [EVSE] Received ChargeParameterDiscoveryReq. Extracting SoC parameters via DC
[127478ms] [EVSE] responding (55bytes) = 01 FE 80 01 00 00 00 2F 80 9A 02 00 40 80 C1 01 41 81 C2 10 80 00 48 00 40 00 00 C0 C3 20 04 0C 0E 01 40 60 A1 84 06 06 06 00 20 60 A1 90 02 03 03 00 50 30 30 05 10 
[127481ms] [EVSE] from 4 entering 4
[SNIFFER] EXI from 49154 to 15118 (36bytes) = 80 9A 02 00 40 80 C1 01 41 81 C2 10 71 90 00 00 08 E0 40 61 10 4E 04 07 0A 8C 30 10 20 50 88 27 12 80 50 00 
[SNIFFER] EXI from 15118 to 49154 (47bytes) = 80 9A 02 00 40 80 C1 01 41 81 C2 10 80 00 48 00 40 00 00 C0 C3 20 04 0C 0E 01 40 60 A1 84 06 06 06 00 20 60 A1 90 02 03 03 00 50 30 30 05 10 
[127519ms] connection closed
[127538ms] [EVSE] re-initializing fsmEvse due to broken connection
[127539ms] [EVSE] re-initializing fsmEvse
[127539ms] Trying to reset the TCP socket
[127548ms] pyPlcTcpSocket listening on port 15118
[129972ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5
[132758ms] [CONNMGR] 9 0 0 0 0 0 0  --> 5

MarkG-PE, 2024-01-16:
I noted in https://github.com/EVerest/logfiles that there is a BMW IX (doesn't say IX3 but assume its the same car modem in any case), that they have charge parameter discovery request on Line 27 and then start cable check on Line 28. 

https://github.com/EVerest/logfiles/blob/main/BMW/iX_fw_11_2022.64/din_spec_70121/2023-03-30T06%3A24%3A26.582Z_din_spec/eventlog.html#L27

In the above log at line 28, it indicates the EVSE initiates to "start cable check"
For some reason with the IX3 in my case, i am not seeing the start cable check, but it just goes straight into broken connection.

uhi22, 2024-01-17:
During the CableCheck, the charger puts voltage on the DC pins, to measure the isolation resistance. In theory, the car could check whether it sees this voltage, and could abort, if it does not see any. But this would be only relevant, if the car would come to CableCheck. My understanding is, that the **BMW ix3** and the **Togg** both are failing during the **ChargeParameterDiscovery**, this is BEFORE the CableCheck is initiated. The ChargeParameterDiscovery is just a "software" step, the connector is still not locked, no voltage is applied to the DC pins.

I propose three actions:
1. Record a TCP dump also from the Togg, to see whether it is exactly the same issue or a different. (Use wireshark or tcpdump to create such a dump. The text log is nice, but does not contain the details about timing and TCP handshake)
2. Evaluate the logs of the iX from everest. Maybe they show a helpful content of the ChargeParameterDiscoveryResponse, and we could adjust the pyPLC / OpenV2Gx to use these better values.
3. Normally, after ChargeParameterDiscoveryRequest and Response are done, the car would lock the connector (to avoid dangerous situations due to high voltage during CableCheck). If the car is not happy with the mechanical connector, it could abort after the ChargeParameterDiscovery. The car may check, whether the PP resistance is ok, also whether the 5% PWM with the correct levels is on the CP, and whether a connector is physically present and whether the lock bolt is able to run into its target position and whether it does NOT run too far due to a missing or damaged connector.

uhi22, 2024-01-17:
Regarding 2: The iX log from everest (...) contain the
ChargeParameterDiscoveryRequest 01fe800100000024809a0211c1ff77aed4fdff907190000009204061104e04070a8c30102050961f12805000
and the ChargeParameterDiscoveryResponse 01fe800100000024809a0211c1ff77aed4fdff90800000040020000405182824138550008000018180c80c1c
So the next step would be, to compare this response to the response from pyPLC/OpenV2Gx (01 FE 80 01 00 00 00 2F 80 9A 02 00 40 80 C1 01 41 81 C2 10 80 00 48 00 40 00 00 C0 C3 20 04 0C 0E 01 40 60 A1 84 06 06 06 00 20 60 A1 90 02 03 03 00 50 30 30 05 10), maybe there is an obvious difference.

MarkG-PE, 2024-01-17:
@Uhi22 i can confirm on item 3 i have tried and tested these issues. I don't believe they are the issue.
My 5% signal was very well formed in rise time and magnitude and duty cycle.
I tested with a plug in the car to ensure that the cars interactions with the plug were handled properly.
The PP resistance i was using is spot on 1.5kOHms so no issues there either.
I think you can eliminate option 3 and i think your post above regarding option 2 is the correct path to follow.
I am happy to implement the test on the IX3 if you can form the matching message in pyPLC??

uhi22, 2024-01-17:
That' the comparision. Surprisingly, the everest sends a much shorter message, containing less parameters, and it has success (or we have an independent issue). I looked to logs of Ionity charger, they send all the data which also pyPLC sends, end even longer messages (68 byte ChargeParameterDiscoveryResponse). Strange, but maybe we just should try the "very short" version of everest to at least see whether it makes a difference.
[image]

uhi22, 2024-01-17:
> I examined it with Wireshark. How did you get the Bmw ix3 chargeparameter response in the picture above?

(Quick and dirty) I copied the exi byte stream from the logs (everest html log and the textlog above in this discussion), added them into the "demo" section of my python exi connector (https://github.com/uhi22/pyPLC/commit/c0bc567f01233df1016c11f20085308810ecfc36) and ran this with "python exiconnector.py". The script calls the openV2Gx, which decodes the content into json. Then copied the decoder results from the command window into two separate text files, and used a diff tool to show the differences of the two files.

But the handwritten command line interface of OpenV2Gx does the decoding only partially, not all fields are implemented in the decoder, only the essential ones. (I'm a little bit lazy ;-) ) So using the wireshark decoder is much better, but requires the pcap. Do you use the Wireshark decoder on Windows? I tried to understand how to install it, but did not get it. A "beginners description" would really help me.

uhi22, 2024-01-17:
And another path: The SAScheduleList indeed may be an issue. I do not have the DIN spec, but at least in the ISO it is marked as "optional", but with the explanation that omitting is only allowed during EVSEProcessing is ongoing. This would mean, that the SAScheduleList is mandatory in our case, when the EVSEProcessing is Finished. If the car strictly checks this, then it is a reason to abort. Will try to push an update.

uhi22, 2024-01-17:
And here we go. Added encoding and some basic decoding of the SAScheduleList in ChargeParameterDiscoveryResponse. Now we have one entry in the schedule, with relative start time 0 (means immediate) and no duration (means endless, I guess). The everest and an other charger which I checked also use relative start time 0, but also set the duration to 86400 (assumingly seconds, is one day). But for the moment let's keep the things simple and let's test whether this already works.
https://github.com/uhi22/OpenV2Gx/commit/42f66dd91b37a44475d1980d8151ed3ec717140d
Does this change anything for BMW and Togg?

Mustafa19981903, 2024-01-17:
@uhi22 Thank you very much for your reply.  I will update the latest version of the main_commandlineinterface.c file.  I will compile OpenV2Gx with its current version and try it in the Togg tool tomorrow.  Especially what should I set the maximum power value for Evse?  It's currently written as 10, which is a value in kW, right?  Should I update the maximum voltage and maximum current values or should I end the compilation process with the old values?

uhi22, 2024-01-17:
The PMax scaling is not clear. The everest sends 10000, so they seem to assume Watts. But the data type is int16, so limited to 32767, which is too small for Watts. So I assumed it is kW, and have set it to 10. On a log of a public charger I saw a negative value, so it is confusing. I just assume that it does not matter.
You could give a try with the sources from github, and depending on the result, do more modifications, maybe reduce the voltage and current.

Mustafa19981903, 2024-01-18:
@uhi22 @MarkG-PE  I made the updates. I will try. I will let you know after I try it.
[image]

uhi22, 2024-01-18:
Somebody patched the PowerLimit to 20 Gigawatts. Multiplier 3 and value 20000, means 20000*10^3 :-)

Mustafa19981903, 2024-01-18:
@uhi22 I fixed it and updated it to 20 again and today I tested it with the togg tool and went to the precharge step, thank you.
I am sharing the document after the ChargeParameter of the Togg tool.
[dc_test.txt](https://github.com/uhi22/pyPLC/files/13977213/dc_test.txt)

uhi22, 2024-01-18:
Great, one issue solved. Hopefully also the BMW is happy now...

Mustafa19981903, 2024-01-19:
@uhi22 What exactly is the difference between Din Spec 70121 and ISO 15118? Does Pyplc software support iso15118?
I had the opportunity to try it with Tesla and there was no iso15118 there. Tesla had a special charging protocol

uhi22, 2024-01-19:
The DIN is a older version then the ISO. Many things seem to be similar, but there are significant differences in message structures. A comparision is difficult, my only reference is the source code of the exi decoder (openV2G).
The pyPLC only implements DIN, not ISO.
If I remember correctly, tesla cars announce in the protocol handshake two protocols, Tesla and DIN. The pyPLC should select DIN in this case.
The voltage is not limited by the protocol technically. Maybe there is somewhere a specified limit of 500V, I do not have the DIN spec. The Car just announces a limit which does not hurt the battery, I guess.

Mustafa19981903, 2024-01-19:
@uhi22 Thank you for your answer.
Can pyPlc software add iso15188? How should we proceed for the addition process?

uhi22, 2024-01-19:
For the ISO15118 discussion, opened a new issue: https://github.com/uhi22/pyPLC/issues/24

uhi22, 2024-01-19:
Found the tesla topic here: https://github.com/uhi22/pyPLC/issues/9
"Vehicle supports 2 protocols. ProtocolEntry#1 ProtocolNamespace=urn:din:70121:2012:MsgDef Version=2.0 SchemaID=1 Priority=2 ProtocolEntry#2 ProtocolNamespace=urn:tesla:din:2018:MsgDef Version=0.7 SchemaID=2 Priority=1 "

Mustafa19981903, 2024-01-23:
@uhi Thank you for the answer, I will try to implement and improve it.

Mustafa19981903, 2024-01-23:
@ After the precharge phase, to move to the power delivery and current demand phase, does the vehicle continue by measuring the dc + and dc- terminals or does it take the evsepresentvoltage value as reference?

uhi22, 2024-01-23:
The end criteria for the precharge phase is, that the chargers voltage has reached the battery voltage with some tolerance. Basically the car has the option to use the EVSEPresentVoltage for this decision, or it can physically measure the inlet voltage, or a combination of both.
It seems not fully specified, what is the intended behavior, so there seems to be some freedom of the car manufacturer to decide which criteria they use. There are some cars, which rely on the EVSEPresentVoltage, and so are moving on to the powerdelivery and currentdemand, without even connecting anything to the DC pins. But other cars seem to be more picky and want to see a physical precharge voltage on the DC pins to finish the precharge phase. Some results are collected here: https://openinverter.org/forum/viewtopic.php?t=3551 If you have results for other cars, just feel free to add it there.

Mustafa19981903, 2024-02-01:
@uhi22 When the vehicle sends target voltage = 450 volts and target current = 100 amps, can I charge values such as 50 volts, 100 volts, 20 amps and 50 amps to the vehicle? Do the vehicles want values close to the target voltage and target current they send?

uhi22, 2024-02-01:
For the voltage, the charger has no choice, the actual voltage is the accu voltage, example 350V. The vehicle sends targetvoltage eg 400V, which only means that if the accu gets fuller, the charger shall not provide more than 400V. So yes, it is a normal case that the presentvoltage is below the target voltage.
For the current, the car says it would be happy with eg 100A. This means, the charger shall limit the current to this value. It is also a normal case, that a small charger provides less current than the car whishes in the targetcurrent.

So basically yes, you can charge with lower values. But not with 50V or 100V, because usual cars have above 250V even battery is empty.

MarkG-PE, 2024-03-30:
@uhi22 The issue is now resolved for me on the BMW iX3.
I downloaded the latest version of pyPlc with SAScheduleList updates and loaded it on the raspberry pi.
I made the connection and it went very quickly in cable pre-charge.

Just to be very transparent also, as i was downloading the pyPlc i noted in your setup instructions there was a section at the start for downloading open-plc-utils to then flash the modem. I did that also and flashed the QC7000 modem as an EVSE. I don't believe that was needed, but i did it anyway. So the raspberry pi had open-plc-utils installed, then pyPlc and openV2Gx.

I noted that latest version of python was 3.11.2 but that was giving me issues with installing things on the raspberry pi, so i reverted it to version 3.9.2 and it was all very straight forward. I captured a wireshark log of it, if you would like it then i can send it to you for record. 

Great Job. !

MarkG-PE, 2024-04-02:
@Mustafa19981903 , I detailed it here
https://github.com/SmartEVSE/SmartEVSE-3/issues/25#issuecomment-1657180723

uhi22, 2024-05-31:
Assuming the original topic is clarified in the meanwhile. Closing.

closed 2024-05-31 by uhi22

uhi22, 2025-07-25:
Please:
- Open a new issue for the new topic.
- describe clearly, on which side (EV or EVSE) you want to change which parameter.

navyareddy1234567, 2025-08-01:
@uhi22 
 please reply