uhi22/pyPLC #15: pyPlc raspberry pi

Fatih738, 2023-09-21:
Im getting error here? How can I solve this

thanks

[image]

Fatih738, 2023-09-21:
Im getting this error. Im using mobaxterm to connect raspberry pi terminal

[image]

Fatih738, 2023-09-21:
I achieved here

But how can I solve this

[image]

Fatih738, 2023-09-21:
I changed all eth0 to eth1.  But Ihow can I solve this

starting in PevMode
initializing pyPlcWorker
[addressManager] we have local MAC CA:14:EA:DE:98:46.
[addressManager] Found 1 link-local IPv6 addresses.
fe80::7744:80c2:58eb:84ee
[addressManager] Local IPv6 is fe80::7744:80c2:58eb:84ee
Linux interface is eth1
[addressManager] will give local MAC CA:14:EA:DE:98:46
[addressManager] will give local MAC CA:14:EA:DE:98:46
pyPlcIpv6 started with ownMac CA:14:EA:DE:98:46
[addressManager] will give local MAC CA:14:EA:DE:98:46
udplog started with ownMac CA:14:EA:DE:98:46
logging to UDP Syslog is enabled
sniffer created at eth1
[427ms] [HARDWAREINTERFACE] Auto detection of serial ports. Available serial ports:
[438ms] [HARDWAREINTERFACE] ignoring /dev/ttyAMA0, because this is not an USB serial port
[439ms] [HARDWAREINTERFACE] ERROR: No serial ports found. No hardware interaction possible.
Traceback (most recent call last):
  File "pyPlc.py", line 133, in <module>
    worker=pyPlcWorker.pyPlcWorker(cbAddToTrace, cbShowStatus, myMode, isSimulationMode)
  File "/home/pi/myprogs/pyPlc/pyPlcWorker.py", line 34, in __init__
    self.hardwareInterface = hardwareInterface.hardwareInterface(self.workerAddToTrace, self.showStatus)
  File "/home/pi/myprogs/pyPlc/hardwareInterface.py", line 263, in __init__
    self.initPorts()
  File "/home/pi/myprogs/pyPlc/hardwareInterface.py", line 220, in initPorts
    self.canbus = can.interface.Bus(bustype='socketcan', channel="can0", can_filters = filters)
  File "/usr/local/lib/python3.7/dist-packages/can/util.py", line 369, in wrapper
    return f(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/can/interface.py", line 138, in __new__
    bus = cls(channel, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/can/interfaces/socketcan/socketcan.py", line 705, in __init__
    bind_socket(self.socket, channel)
  File "/usr/local/lib/python3.7/dist-packages/can/interfaces/socketcan/socketcan.py", line 512, in bind_socket
    sock.bind((channel,))
OSError: [Errno 19] No such device
SocketcanBus was not properly shut down

Fatih738, 2023-09-22:
I have succesfully completed the charging station with pyPlc on raspberry pi 4. 

Best wishes. thanks

uhi22, 2023-09-22:
Could you please summarize your "Lessons learned"? Something like: "I had the problem xyz. The root cause was this and that. I fixed it by doing this and that.", for each or your topics above. This could be a big help for others who run into the same troubles.

Fatih738, 2023-09-22:
Actually, looking at the step I've already started, you've already made everything clear. The parts that I had to change when applying: Making all eth0s eth1, installing pcan driver, pev mac address. Now I'm looking at the parts where I send charging parameters from bms, read the cp and pp pins as hardware. I'm going to test the Canbus communication. I'll look at receiving and sending data. I had to use real vnc and touchscreen due to localhost 10.0 error to run the code. I usually use mobaxterm. This is how it was solved, but maybe there is another solution. I will try to run with .service file.

Fatih738, 2023-09-28:
I have added BMS section. SOC, current, voltage. I want to go to the dc fast charging station. I m going to charge my car with pyPlc. Can I do that. please confirm me. I m going to charge with 5A, 10A. just test.

Fatih738, 2023-10-02:
Is pyPlc read the cp duty?

uhi22, 2023-10-02:
No, it does not care for the CP duty. Neither on evse nor on pev side.

Fatih738, 2023-10-02:
I have to implement the cp duty. Do you know any application running on raspberrypi? how can we read duty with raspberry pi? Also today I have gone to dc charging station and charging session is started and completed succesfully.

Where did come 398V, 100A in code?

thanks

uhi22, 2023-10-02:
I never did duty cycle measurements with a raspberry. Maybe this feature is supported in hardware by a timer unit, maybe not. Or per pin-interrupt, not sure whether this is fast enough. Maybe only possible in kernel mode.

The the 398V and 100A are just hard coded in the encoder of the ChargeParameterDiscoveryRequest, e.g. here: https://github.com/uhi22/OpenV2Gx/blob/d0e22816ace99ff77b72bcb45273580ea65211a9/src/test/main_commandlineinterface.c#L1275

Fatih738, 2023-10-04:
today I have tested with dc charge station again. I couldnt make the slac match. We successfully completed all the charging pocess on Monday. sometimes I saw ATTEN_CHAR.IND ATTEN_CHAR.IND ATTEN_CHAR.IND. 

pyPlc.py is running on raspberry pi. are mac adress or another thing what they're about?

Do you have opinion about this.

uhi22, 2023-10-04:
Without a log of the communication it is difficult to guess. Make an ethernet log, with Wireshark or tcpdump. Celerons startup scripts are nicely creating tcpdump recordings per session, I think this is described in the readmes.

Fatih738, 2023-10-04:
this is a my test record. codico vw-carrier-board e is evse side. pyPlc is pev side.
[2023-10-04_132212_pevNoGui.log](https://github.com/uhi22/pyPLC/files/12803591/2023-10-04_132212_pevNoGui.log)

uhi22, 2023-10-04:
Just checked the log, searching for the issue, but then noticed: There is none :-D Only the end seems missing, looks like aborted. Wish much success on the real charger.

Fatih738, 2023-10-05:
our log record

slac matcişng failed. how can we solved this

[2023-10-05_074645_pevNoGui.zip](https://github.com/uhi22/pyPLC/files/12813949/2023-
10-05_074645_pevNoGui.zip)

Fatih738, 2023-10-05:
I have done all process again with pyPlc on other raspberrypi

two codes running and no slac matcing

actually I did same as before

[image]

uhi22, 2023-10-05:
Changing the pevmac and NID should not be necessary, the pyPLC takes the actual MAC from the Raspberry. In the pcap we see PEV MAC a2:c9:c5:ff:8b:98 and the modem has CIMSYS_33:44:55 (00:11:22:33:44:55). We see also, that the modem is correctly detected, it sends a SoftwareVersionResponse. The pyPLC sends slacParamRequests, but never receives a response for this. So I see no configuration error on the pev side.

Did you use the identical setup on the public charger? This would confirm, that it is fine.
Some ideas for further investigations:
- Is the CP connection to the Codico-EVSE correctly connected?
- Is the Codico-EVSE working with an other PEV?
- Do you observe on the CP line coordinator packages sent by the EVSE side (this would be "tock" sounds, multiple per second, when using https://github.com/uhi22/pyPLC/blob/master/doc/hardware.md#ccs-detector, or oscilloscope)?
- Do you observe on the CP line the SLAC packages (single "tock" sound, in addition to the coordinator packages above)?

Maybe there is a configuration issue or software issue with the Codico EVSE. Not sure what it wants to say with the log "EV is already connected."

Fatih738, 2023-10-05:
"EV is already connected." means diode and 2.7k resistor are detected according to IEC61851. I have diode and resistor. Also I have added 1.3k resistor and switch for stateB and StateC. 

when I went to the station, charge didnt start with charging station. But I changed CP and PE to codico EVSE board, slac matching successfull and charging process was completed. again I changed CP and PE to charging station, it didnt.

Actually I succesfully completed charging session on monday. tuesday, wednesday and today, charging is failed.. 
When I did succesfully on monday, I didnt change the code until tuesday. When I went to charging station again, failed. I didnt understand why

1-Is the CP connection to the Codico-EVSE correctly connected?
yes
2-Is the Codico-EVSE working with an other PEV
codico EVSE is wb-carrier-board-e. just I have tested with pyPlc. If you mean this
3-Do you observe on the CP line coordinator packages sent by the EVSE side (this would be "tock" sounds, multiple per second, when using https://github.com/uhi22/pyPLC/blob/master/doc/hardware.md#ccs-detector, or oscilloscope)?
ok. I will test
4-Do you observe on the CP line the SLAC packages (single "tock" sound, in addition to the coordinator packages above)?
thanks. I will test

Fatih738, 2023-10-05:
Sometimes this happened. When I start again the EVSE side, problem solves. I didnt take a log 

[image]

Fatih738, 2023-10-06:
do you use burs mode or latency mode?

uhi22, 2023-10-06:
The foccci board with ccs32clara software uses burst mode. But this is not related to pyPLC and raspberry. I have no idea what the linux driver of the raspberry uses, and how the codico board would choose the one or the other.

Fatih738, 2023-10-11:
battery dc + dc- are connected over two contactors to charging station. When I can open and close the contactors in code?
Also what are the "setPowerRelayOn" "setPowerRelayOff"  "setRelay2On" "setRelay2Off" functions.  What is the duty of relay2 in setrelay2on.

uhi22, 2023-10-11:
Not sure whether I got the question. The two power contactors for DC plus and DC minus are named powerrelay and relay2 in code.

Fatih738, 2023-11-11:
Can I add different task as thread in pyPlc.py or hardware interface section.
Actually I want to combine different python codes. How to I combine different python codes with pyPlc

Fatih738, 2023-11-14:
Can I add different task as thread in pyPlc.py or hardware interface section.
Actually I want to combine different python codes. How to I combine different python codes with pyPlc
Can we run simulataneously different python codes that different directory in starter.sh file?

thanks a lot,

best wishes

uhi22, 2023-11-14:
The pyPLC itself is not designed to support multiple threads. It is written to be easily portable to C on microcontroller, using a simple scheduler. There are concepts in python to support multithreading, but I personally never used them. You could try to port the pyPLC to multithreading. Or run a separate python instance for the other application, and let they talk to each other via local tcp or http or something like this.

Fatih738, 2023-11-23:
can I add mcp2551 to code. mcp 2551 works with spi you know and qca use spi line

closed 2023-11-29 by Fatih738