uhi22/pyPLC #16: pyPLC on Raspberry4 with OS lite

GabrielAtlasPower, 2023-10-26:
Hi guys! 

Thanks for your hard work and great advances with this project. 

I'm trying to run "pyPLC.py E S" in a raspberry without desktop and getting this message:

`starting in EvseMode, simulated environment
Traceback (most recent call last):
  File "/home/pi/myprogs/pyPlc/pyPlc.py", line 96, in <module>
    root = tk.Tk()
           ^^^^^^^
  File "/usr/lib/python3.11/tkinter/__init__.py", line 2326, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
_tkinter.TclError: no display name and no $DISPLAY environment variable
`
Do you faced this problem before? How can I run using this lite OS? 

Thanks.

uhi22, 2023-10-27:
The pyPlc.py is the variant which is using the graphical user interface, with tkinter. This does not work on a display-less device. Instead, use the pevnogui or evsenogui variants. But I'm not sure whether they support the simulation mode.

GabrielAtlasPower, 2023-10-27:
Ok, no problem. Now I'm testing on Windows.

closed 2023-11-07 by GabrielAtlasPower