uhi22/pyPLC #17: PEV mode connection break after change the deltaSoc

rossleefmt, 2023-11-06:
I have successfully connected the pyPlc in PEV mode to my EVSE and completed the charging process without any issues. However, when I modify the code to change the deltaSoc in hardwareinterface.py to extend the charging period, the charging process consistently stops at around 30% SoC.

Below is the code I've modified:

[image]

And here is the error log:(57213 is my EVSE)

> [80576ms] [PEV] {
"msgName": "CurrentDemandRes",
"info": "39 bytes to convert", 
"error": "",
"result": "",
"schema": "DIN",
"g_errn": "0",
"header.SessionID": "76ef8536fc30cafd",
"header.Notification_isUsed": "0",
"header.Signature_isUsed": "0",
"ResponseCode": "OK",
"DC_EVSEStatus.EVSEIsolationStatus": "0",
"DC_EVSEStatus.EVSEIsolationStatus_isUsed": "1",
"DC_EVSEStatus.EVSEStatusCode": "1",
"EVSEStatusCode_text": "EVSE_Ready",
"DC_EVSEStatus.NotificationMaxDelay": "0",
"DC_EVSEStatus.EVSENotification": "0",
"EVSEPresentVoltage.Multiplier": "0",
"EVSEPresentVoltage.Value": "398",
"EVSEPresentVoltage.Unit": "V",
"EVSEPresentCurrent.Multiplier": "2",
"EVSEPresentCurrent.Value": "1",
"EVSEPresentCurrent.Unit": "A",
"EVSECurrentLimitAchieved": "1",
"EVSEVoltageLimitAchieved": "0",
"EVSEPowerLimitAchieved": "1",
"EVSEMaximumVoltageLimit.Multiplier": "2",
"EVSEMaximumVoltageLimit.Value": "4",
"EVSEMaximumVoltageLimit.Unit": "V",
"EVSEMaximumCurrentLimit.Multiplier": "2",
"EVSEMaximumCurrentLimit.Value": "1",
"EVSEMaximumCurrentLimit.Unit": "A",
"EVSEMaximumPowerLimit.Multiplier": "3",
"EVSEMaximumPowerLimit.Value": "25",
"EVSEMaximumPowerLimit.Unit": "W",
"debug": "Line442"
}
[80621ms] [PEV] responding (43bytes) = 01 FE 80 01 00 00 00 23 80 9A 02 1D BB E1 4D BF 0C 32 BF 50 D1 40 06 80 C0 C1 90 60 01 81 05 80 48 0C 08 36 
01 00 C1 42 38 0C 00 
[80621ms] [PEV] from 14:WaitForCurrentDemandResponse entering 14:WaitForCurrentDemandResponse
[SNIFFER] EXI from 57213 to 52880 (39bytes) = 80 9A 02 1D BB E1 4D BF 0C 32 BF 50 E0 00 00 80 00 01 82 84 70 18 28 18 00 84 04 0A 0A 00 80 50 30 01 
03 03 80 C8 00 
[81023ms] [CONNMGR] 9 0 0 0 0 0 318  --> 100
[SNIFFER] EXI from 57213 to 52880 (39bytes) = 80 9A 02 1D BB E1 4D BF 0C 32 BF 50 E0 00 00 80 00 01 82 84 70 18 28 18 00 84 04 0A 0A 00 80 50 30 01 03 03 80 C8 00 
[SNIFFER] EXI from 57213 to 52880 (39bytes) = 80 9A 02 1D BB E1 4D BF 0C 32 BF 50 E0 00 00 80 00 01 82 84 70 18 28 18 00 84 04 0A 0A 00 80 50 30 01 03 03 80 C8 00 
[82164ms] [CONNMGR] 9 0 0 0 0 0 285  --> 100
[83297ms] [CONNMGR] 9 0 0 0 0 0 252  --> 100
[SNIFFER] EXI from 57213 to 52880 (39bytes) = 80 9A 02 1D BB E1 4D BF 0C 32 BF 50 E0 00 00 80 00 01 82 84 70 18 28 18 00 84 04 0A 0A 00 80 50 30 01 03 03 80 C8 00 
[84441ms] [CONNMGR] 9 0 0 0 0 0 219  --> 100
[85565ms] [CONNMGR] 9 0 0 0 0 0 186  --> 100
[86297ms] [PEV] from 14:WaitForCurrentDemandResponse entering 99:SequenceTimeout
[86353ms] [PEV] Safe-shutdown-sequence: setting state B
[86353ms] [HARDWAREINTERFACE] Setting CP line into state B.
[86353ms] [PEV] from 99:SequenceTimeout entering 111:SafeShutDownWaitForChargerShutdown

It appears that when the issue occurs, the PEV has sent a "CurrentDemandRes" message, but the [SNIFFER] is unable to capture it as it normally would. Are there any known issues that could cause this problem, or is there something incorrect with the code modification?

uhi22, 2023-11-06:
The modification is fine. An ethernet log using wireshark or tcpdump would possibly help.

rossleefmt, 2023-11-06:
The tcpdump file is attached for reference.
[1107_0840.zip](https://github.com/uhi22/pyPLC/files/13273306/1107_0840.zip)

uhi22, 2023-11-15:
Seems there is some instance which is resetting the ongoing TCP connection. I think we need an linux networking expert to explain and solve this.
[image]

rossleefmt, 2023-11-17:
I've done some tests below and the result indicated that it's not caused by the Multicast Listener message.

Firstly I disabled the Multicast Listener message on Raspberry Pi. 
Then I run pyPLC in EVSE and PEV mode, both set "eth_interface = eth0"
This time there is no ICMP messages but still the EVSE fail to receive the last message from PEV.

[image]

rossleefmt, 2023-12-01:
After we changed the RPi version to v11, the problem was solved. Previously we used RPi v12.

closed 2023-12-01 by rossleefmt