uhi22/pyPLC #18: EVSE MODE CANBUS ERROR

Mustafa19981903, 2023-11-14:
Hello, even though I follow the installation steps separately for Raspberry Pi and Windows, I get such an error. How can I solve this error? Thank you.
[image]

uhi22, 2023-11-14:
Do you plan to use CAN bus on Windows machine? If yes, we need to add a proper interface layer. The currently used socketCAN is linux specific. If you do not want the CAN, just adjust the pyPLC.ini file, set charge_parameter_backend = none instead of ChaDeMo, this removes the CAN dependencies.

Mustafa19981903, 2023-11-15:
[image]
[image]

Mustafa19981903, 2023-11-15:
It came out like this, I bought it, what can I do? My car is Zoe.

Mustafa19981903, 2023-11-15:
CanBus Error Solved.

closed 2023-11-15 by Mustafa19981903