uhi22/pyPLC #19: DC Electric Vehicle Charging

pamukcu09, 2023-11-15:
Hi,

Can I only read data from the vehicle in this project? 

When I run this software with my PWM signal with 5% Duty ratio, can I switch it to State C for DC Charging?

uhi22, 2023-11-15:
I understand you plan to use pyPLC as EVSE. In this case, the car will switch from StateB to StateC (see https://github.com/uhi22/pyPLC#example-flow checkpoint 550). The current experience is, that pyPLC is able to convince the car to close the contactors, so basically it is possible to really charge the vehicle, not only to read out data.
In case of more detailled questions, please always mention the checkpoint number in https://github.com/uhi22/pyPLC#example-flow , together with your expected behavior and the observed behavior.

uhi22, 2024-01-19:
Question seems to be answered. Closing.

closed 2024-01-19 by uhi22