uhi22/pyPLC #22: PLC MODULE HARDWARE PROBLEM

Mustafa19981903, 2023-11-20:
I am using the tl-pa4010p module as PLC and I could not establish a connection with the vehicle. Can you recommend a plc module that can work with the software?

Mustafa19981903, 2023-11-20:
How did you set these values in the software file with the extension pyplchomeplug.py? I will be very pleased if you give me information.
[image]

Mustafa19981903, 2023-11-20:
[image]

pamukcu09, 2023-11-22:
> TL-pa4010p modülünü PLC olarak kullanıyorum ve araçla bağlantı kuramadım. Yazılımla çalışabilecek bir plc modülü önerebilir misiniz?

@uhi22

pamukcu09, 2023-11-22:
> I am using the tl-pa4010p module as PLC and I could not establish a connection with the vehicle. Can you recommend a plc module that can work with the software?

@uhi22

uhi22, 2023-11-22:
The tl-pa4010p which I used are working. To find out what happens it would be helpful that you add a description what you did, what you expect and what you observe. In best case together with an ethernet log, which you can record using wireshark or tcpdump.

pamukcu09, 2023-11-23:
> [image]

We just wondering, if we use different plc module do we have to change this parameters? @uhi22

uhi22, 2023-11-23:
These values come from the "official" qualcom QCA tool on GitHub, and from the public homeplug documents. If you use plc hardware from other than qualcom/atheros maybe they have different requirements.

pamukcu09, 2023-11-23:
Thank you so much, we are appreciate for your help

closed 2024-01-16 by Mustafa19981903