uhi22/pyPLC #23: triggerConnectorLocking & triggerConnectorUnlocking

Mustafa19981903, 2023-12-12:
@uhi22 
Hello again UHI
I'm thinking of making changes on the hardware side.
I need to look at fsmPev.py for the charging station software because the Response commands are there. FsmEvse.py Request commands appear.
Do the connector unlocking and closing functions unlock the connectors inside the vehicle or do they perform an operation at our charging station?

uhi22, 2023-12-12:
Hi,
you write "look at fsmPev.py for the charging station software". This is a misunderstanding. The concept is:
- The PEV (Plug-in-electrical-vehicle) is the car.
- The EVSE (EV Supply Entrity) is the charger.
This means, if you are developing a charger, you will adapt/extend/improve the chargers finite state machine, fsmEvse.py.
The connector lock in CCS case is a pure vehicle thing. The vehicle knows, when to lock and unlock, and is responsible to do this. The motor of the connector lock is driven by the vehicles charge controller (fsmPev.py in case of pyPLC). So if you are developing a charger, you do not have to care for connector locking at all.
But wait, there is one relation also for the charger. In https://github.com/uhi22/pyPLC#example-flow, checkpoint 560, the charger receives from the car the information, whether the connector is locked. This is to prevent the high-voltage-cable check in case the connector lock does not work.

Mustafa19981903, 2023-12-12:
@uhi22  thank you for your answer

uhi22, 2023-12-12:
Let me make an example. The fsmPev sends a CurrentDemandRequest via the powerline communication to the fsmEvse. The currentDemandRequest contains the settings of voltage and current, which the car (or more precise: the cars battery management system BMS) thinks is good for the battery. The fsmEvse should take the voltage and current values and forward it to the power supply unit, in your case maybe via CAN to the "30 kW power module".
So you need to find a CAN driver (hardware and software) for your machine, which could could use together with python. On linux, there is socketCAN integrated into the kernal, even with a driver for the MCP2515 CAN transceiver. There should be phython libraries for use with SocketCAN. So your way would be to write some code in the fsmEVSE handleCurrentDemandRequest function, which takes the voltage and current from this message and sends it via CAN.

Mustafa19981903, 2023-12-13:
@uhi22 Thank you for your answer. 
I have two more questions.
As far as I understand, we create json data by processing the hexadecimal data of the rxdata array, how do we do this?
I bought two tplink-pa4010 plc modules and wanted to power them from the dc side and receive slac messages, but I was not successful. I think I've made a hardware mistake. I couldn't find a photo with a clear link. I ran the simulation software with both Windows and Raspberry Pi. But I couldn't run it in real time. I could not establish a connection between 2 raspberry pi and 2 plc modules. Later, I provided 5% duty from the 12 volt 1 khz circuit with the card of my own design, but I still could not receive the slac messages with the vehicle. I will buy 2 new plc modules. I would be very happy if you can support me on how to connect them exactly.
[image]
[image]

Mustafa19981903, 2023-12-13:
@uhi22 A special configuration was required for the plc module to transmit slac messages. I saw the steps to be followed for this, but I did not fully understand it. Do I need to connect Raspberry Pi to the plc module via ethernet and write commands at http://github.com/qca/open-plc-utils?

uhi22, 2023-12-13:
Short answer: yes, you install the qca tool from this link on your raspberry, connect it via ethernet cable to the adapter and run the read/patch/write commands on the raspberries command shell. Long answer: there are a lot of html pages with explanations included in the qca tool.

uhi22, 2024-01-19:
Question seems to be answered, closing.