uhi22/pyPLC #25: Detail information for constructing circuit

pezluna, 2024-02-14:
Hello.

I used a translator, so my wording may be strange.

I am currently working on an EV related project in graduate school and I am very interested in this project.

I'm trying to construct the circuit myself based on the pictures attached to all the markdown, but I'm having trouble identifying the circuit components correctly.

1. if you don't mind me asking, is it possible to get the circuit diagram drawn through the software for everyone who is interested in this project and wants to try to construct the circuit?

2. I would like to purchase the components for the circuit, but the information is scattered in so many different markdown files, is it possible to list up the components used in the circuit?

uhi22, 2024-02-14:
Hello and welcome.

First: The translation is quite good, at least for somebody like me, that is not native speaker anyway :-)

The components which are required heavily depend on what you are planning to do. Try to answer these questions for yourself:
1. Do you plan to build the car side or the infrastructure side, or both?
2. Are you interested only in the communication, or do you plan that real energy shall flow?
3. What interfaces do you plan to the "outside world"?
4. What kind of hardware do you plan to use (e.g. windows PC, or Raspberry or microcontroller)?
Depending on the answers, the needed components are quite different.
5. Do you just plan some experiments on the desk, or do you plan to interact with real car or real charging station?

pezluna, 2024-02-15:
Thank you for your quick reply. 
I apologize for my lack of clarification, so I'll answer your 5 questions.

1. Currently, there are no PEVs and EVSEs available for experiments, so I would like to build both if possible. 
2. I'm currently interested in communication, but it would be nice to have actual energy flowing.
3. I don't exactly understand what you mean by interfaces about outside world. If you mean a debug interface like USB-to-Serial, the answer is yes.
4. For further experimentation, I think it would be best to utilize a Raspberry Pi.
5. I will be building all possible communication environments at my desk and plan to interact with a real charging station as a further experiment.

Currently, I have TP-Link TL-PA4010P, Arduino Pro Mini and Raspberry Pi.

uhi22, 2024-05-31:
Assuming everything is clearified in the meanwhile. Closing.

closed 2024-05-31 by uhi22