uhi22/pyPLC #26: Viewing Sent and Received Messages Together in Evse Mode

Mustafa19981903, 2024-03-06:
Hello;
When Pyplc operates in evse mode, only the message packets sent by the electric vehicle are seen. What can I do to see both the message packets sent by the vehicle and the message packets sent by the charger when I run Pyplc in EVSE mode?

For example;
"supportedAppProtocolReq",
"supportedAppProtocolRes",

uhi22, 2024-03-06:
But, the supportedAppProtocolReq is a special case which is not supported by the above update, also not by the pcapConverter.py. Would need improvements.

Mustafa19981903, 2024-03-07:
@uhi22 thank you for the answer. I updated Fsmevse. I received all the messages up to precharge except the Supported App Protocols message.
[log.txt](https://github.com/uhi22/pyPLC/files/14523156/log.txt)

Mustafa19981903, 2024-04-24:
This process is completed.

closed 2024-04-24 by Mustafa19981903