uhi22/pyPLC #28: Checkpoint 535

Mustafa19981903, 2024-06-27:
@uhi22  https://github.com/uhi22/pyPLC#example-flow  I want to add an RFID or similar control at checkpoint 535, 536 and 538.
msg = addV2GTPHeader(exiEncode("EDl")) # EDl for Encode, Din, ContractAuthenticationResponse how do I update this line of code

uhi22, 2024-06-27:
At the moment, the EXI encoder does only support one type of ContractAuthenticationResponse: it says "authentication done". To support the RFID or similar check, we need two different cases, the "done" and the "ongoing" case.
This needs:
* In the exi encoder (OpenV2Gx), between init_dinContractAuthenticationResType() and prepareGlobalStream(), set the field EVSEProcessing to dinEVSEProcessingType_Finished or dinEVSEProcessingType_Ongoing, depending on the first command line parameter, by calling getIntParam(0).
* In pyPLC, add a parameter to the exiEncode("EDl"), something like exiEncode("EDl_"+strAuthFinished), where strAuthFinished is a string with eigther "0" or "1" depending on the result of the RFID check.

Mustafa19981903, 2024-07-02:
**myprogs code update** 
void encodeContractAuthenticationResponse(void) {
    init_dinMessageHeaderWithSessionID();
    init_dinBodyType(&dinDoc.V2G_Message.Body);
    dinDoc.V2G_Message.Body.ContractAuthenticationRes_isUsed = 1u;
    init_dinContractAuthenticationResType(&dinDoc.V2G_Message.Body.ContractAuthenticationRes);
    prepareGlobalStream();
	
	int authStatus =getIntParam(0);
	if(authStatus==1){
		EVSEProcessing=dinEVSEProcessingType_Finished;
	}
	else{
		EVSEProcessing=dinEVSEProcessingType_Ongoing;
	}

	EVSEProcessing=      getIntParam
    g_errn = encode_dinExiDocument(&global_stream1, &dinDoc);
    printGlobalStream();
    sprintf(gInfoString, "encodeContractAuthenticationResponse finished");
}

**fsmevse code uptade**
            if (strConverterResult.find("ContractAuthenticationReq")>0):
                if self.rfid_check_successful:
                    self.strAuthFinished="1"
                else:
                    self.strAuthFinished="0"
                # todo: check the request content, and fill response parameters
                msg = addV2GTPHeader(exiEncode("EDl"+self.strAuthFinished))

@uhi22 Are there any missteps?

uhi22, 2024-07-02:
Looks good, only the underline is missing after the EDI, to separate the parameter.

Mustafa19981903, 2024-07-09:
@uhi22  I am experiencing this issue every time I run the software. When I unplug and replug the Ethernet cable, the software continues to work. I have also changed my IP address within the software, but this issue persists.
starting in EvseMode, simulated environment
initializing pyPlcWorker
[addressManager] we have local MAC D8:3A:DD:7C:4B:59.
[addressManager] Found 0 link-local IPv6 addresses.
[addressManager] Error: No local Ipv6 address was found.
Exiting, because it does not make sense to continue without IPv6 address

uhi22, 2024-07-09:
What is your plan? If you want to run pyPLC against an other machine, the precondition is that an active ethernet connection to the modem is present and the operating system cared for assigning an IP address. Loss of IP adress should be checked in the OS (ipconfig or ifconfig or such things).
If you want to run two instances of pyPLC at the same machine, without ethernet connected, you could try to comment-out the check in addressmanager, I'm not sure whether this solves the issue. Did not use this mode since a long time.

Mustafa19981903, 2024-07-10:
@uhi22  I encounter this error when I run Evse and Pev mode in simulation mode, and when I remove and plug the ethernet cable back in, the problem disappears.
When the simulation software runs, after a certain period of time, it falls into the error from 14:WaitForCurrentDemandResponse entering 99:SequenceTimeout and the simulation charging process ends

Mustafa19981903, 2024-07-16:
****FSMEVSE
       if (strConverterResult.find("ContractAuthenticationReq")>0):
                # todo: check the request content, and fill response parameters
                self.value1, self.value2 = self.hardwareInterface.read_registers()
                #print(self.value1)
                if(self.value1==4):
                    self.strAuthFinished="1"
                    #print("1")
                else:
                    self.strAuthFinished="0"
                   # print("0")
                msg = addV2GTPHeader(exiEncode("EDl_"+self.strAuthFinished)) # EDl for Encode, Din, ContractAuthenticationResponse
                if (testsuite_faultinjection_is_triggered(TC_EVSE_ResponseCode_SequenceError_for_ContractAuthenticationRes)):
                    # send a ContractAuthenticationResponse with Responsecode SequenceError
                    msg = addV2GTPHeader("809a021a3b7c417774813310c0A200")
                self.addToTrace("responding " + prettyHexMessage(msg))
                self.showDecodedTransmitMessage(msg)
                self.publishStatus("ContractAuthentication")
                self.Tcp.transmit(msg)
                self.enterState(stateWaitForFlexibleRequest) # todo: not clear, what is specified in DIN

*****OpenV2Gx

    int authStatus =getIntParam(0);
     if(authStatus==1){
          dinDoc.V2G_Message.Body.ContractAuthenticationRes.EVSEProcessing=dinEVSEProcessingType_Finished;
     }
    else{
         dinDoc.V2G_Message.Body.ContractAuthenticationRes.EVSEProcessing=dinEVSEProcessingType_Ongoing;
    }

I communicated with the fsmevse software with a different processor from the outside with uart. If the incoming value is 4, "EVSEProcessing": "Finished". If the incoming value is not 4, "EVSEProcessing": "Ongoing",. You can make changes to the software according to your need. It was tested with simulation software. The logs are in the file below. This control software only works with din iso15118 how do we integrate?

[log.txt](https://github.com/user-attachments/files/16247859/log.txt)

uhi22, 2024-07-23:
Great, thanks. I now added the authentication feature for both, DIN and ISO1, with these commits:
https://github.com/uhi22/pyPLC/commit/1171d34d280f61286c63d83cb3a01c685985fe46
https://github.com/uhi22/OpenV2Gx/commit/0ee2d2a019bc2a445db59a9456f2706eea59d474
For demonstration I just used a counter to give the response "ongoing" on the first try, and "finished" at the second try. For real-life usage, the function isUserAuthenticated() needs to be extended with the real RFID check (or whatever).
I tested only the DIN so far, so ISO1 is untested at the moment.

uhi22, 2025-01-17:
Seems to be done