uhi22/pyPLC #31: Encountering issues with V2G communication with Tesla

hycm0000, 2024-08-02:
We are currently working on a charging station project, and the test subject is a Tesla Model Y from the United States. 
The V2G communication data is as follows.
Based on the data, this car supports both the 'urn:din:70121:2012:MsgDef' and 'urn:tesla:din:2018:MsgDef' protocols.We have tested both protocols.

1. When the 'urn:tesla:din:2018:MsgDef' protocol is selected, after we send the supportedAppProtocolRes, the car also sends an ACK indicating that the data has been successfully received. However, there are no subsequent V2G requests."

2. When the 'urn:din:70121:2012:MsgDef' protocol is selected, the V2G communication progresses to 'ServiceDiscoveryRes'. After we send the ServiceDiscoveryRes, there are no subsequent V2G requests."

We have been stalled at this point for a long time and hope to get some help，Thank you very much！
[logs.zip](https://github.com/user-attachments/files/16468005/logs.zip)

hycm0000, 2024-08-02:
urn:din:70121:2012:MsgDef：

[image]

urn:tesla:din:2018:MsgDef：

[image]

uhi22, 2024-08-05:
For the tesla schema, we do not know how this works, and also do not have a matching exi decoder.
For the DIN, we had success with multiple european Model Y and pyPLC. So maybe your car expects a different EnergyTransferType, depending on the installed socket.

M4GNV5, 2025-01-08:
In the `ServiceDiscoveryRes` packet the charging station tells the vehicle the `EnergyTransferType`. For the newer standards this allows to switch between AC and DC. For the old DIN SPEC everyone in europe uses `DC_extended` (i.e. the CCS2 connector).

Since your american Tesla does not like your `ServiceDiscoveryRes` maybe the NACS port requires a different value than DC_extended. [The EcoG repo lists possible values](https://github.com/EcoG-io/iso15118/blob/68d810826e725c1939535f2673e8c7bf7acaf158/iso15118/shared/messages/enums.py#L72), but not sure if any of those is correct.
On the other hand with the transition from CCS1 to NACS in america, CCS1 models seem to support NACS charging stations using a plain plug adapter. If Superchargers would advertise a different `EnergyTransferType` CCS1 cars should not be able to charge there :thinking:

closed 2026-02-25 by uhi22