uhi22/pyPLC #35: Status: TCP conn broken (Help)

breezelunar, 2024-11-22:
[BMW-Logs.pdf](https://github.com/user-attachments/files/17869370/BMW-Logs.pdf)
[MINI-COOPER-LOGS.pdf](https://github.com/user-attachments/files/17869371/MINI-COOPER-LOGS.pdf)

Hi uhi22

I have been testing the pyPLC code with a pi5, a PLC Stamp Micro and two vehicles, a MINI Cooper SE and a BMW ix3

I suspect the EV is suspending communication during the supportedAppProtocolReq/-Res, I was hoping you might be able to advise on what is causing the TCP connection to break.

I appreciate your work and your time, thank you in advance.

-breezelunar

breezelunar, 2024-11-24:
Hi uhi22, thank you for the swift reply. Initially I thought it could be the CP line. But the CP line has been tested rigorously with an oscilloscope, I don't believe the physical connection is the issue. I have also disabled the network manager for our specific eth1 port. Any other ideas?

breezelunar, 2024-11-24:
It seems to happen at the same place every time, the supportedAppProtocolReq/-Res. Could this be a clue?

uhi22, 2024-11-24:
We would need the support of a linux expert who knows how view the reason why a TCP connection just closes. First interesting point is a wireshark log of the interface, this should show whether the car or the raspberry closes the TCP.

breezelunar, 2024-11-24:
Thanks uhi, I will check out that wireshark log

breezelunar, 2024-11-26:
Could this be the fix? Let me know

https://github.com/uhi22/pyPLC/issues/9

breezelunar, 2024-11-26:
Does this possibly relate to issue #9 at all?

https://github.com/uhi22/pyPLC/issues/9

breezelunar, 2024-11-26:
Does this possibly relate to issue https://github.com/uhi22/pyPLC/issues/9 at all?

uhi22, 2024-11-26:
At 6.14s there is a FIN, this is the point where the TCP connection is closing.

uhi22, 2024-11-26:
Could you attach the real pcap file?

breezelunar, 2024-11-26:
[capture.pcap.zip](https://github.com/user-attachments/files/17919147/capture.pcap.zip)

breezelunar, 2024-11-28:
Thank you very much uhi for the analysis. Based on the information you provided, I am going to do some work to solve the issue, starting with the PWM on the CP line. Will let you know my findings. Thanks again for your time

lewurm, 2024-12-06:
I had a similar issue with my car (Tesla Model 3).  My problem was indeed the CP line. There is a state change from B to C somewhen soon after the schema selection (i.e. 9V to 6V). That messed with my (faulty) circuit, the solution was to swap the LF353 with a genuine one; you could see with the scope that the slew rate wasn't proper before. The car even gave  somewhat useful errors in its service menu.

breezelunar, 2025-01-09:
lewurm, what did you use as a replacement for the LF353?

uhi22, 2025-01-09:
My understanding is that he used a "faked" Chip or an other type before, and changing to a real LF353 solved the issue.

lewurm, 2025-01-10:
That's right, I used fake LF353 from Aliexpress before. Using a genuine TI LF353 fixed my issue around broken connections after schema selection (that is, the voltage levels on CP looked better then in state C).

uhi22, 2025-01-10:
So my conclusion: If something does not work, take an oscilloscope to measure the CP. This should show nice rectangles of the 1kHz, 5% PWM, with the low voltage nearly at -12V, and the high voltage near 9V (in state B) or later near 6V (in state C).

breezelunar, 2025-01-10:
Yes it seems the chargepump I was using to make the negative rail could be the issue. It only reaches -11.8V. So I am working on a new design now. Otherwise the PWM looks good. No noise, very clean. So will keep updated. Thanks again for the help

breezelunar, 2025-01-10:
I get 12.1V on the positive and about -10.5V on the negative (these are the scope measurements, RIGOL DS1104), I measure -11.8V on the output of the chargepump, so there is even more loss for some reason on the negative level of the CP. 

I have a new circuit now, using MUN5312s and BSS8402s for the 12V/-12V rails.

uhi22, 2025-01-10:
Good point. The nominal voltages are -12V and +9V in state B. And the reality on my setup is -10V and +7V. So two volts are missing in each direction. Nevertheless this worked on some cars. A better (output rail to rail) amplifier would be great, to be fully in spec.
[image]

breezelunar, 2025-01-10:
That's very interesting that even with a negative rail of -10V you are still able to communicate

uhi22, 2025-01-10:
Some people are even saying that some onboard chargers are even happy when using 0V as lower voltage, so just applying a 0V / 12V PWM :-)

breezelunar, 2025-01-10:
[image]

breezelunar, 2025-01-10:
> Some people are even saying that some onboard chargers are even happy when using 0V as lower voltage, so just applying a 0V / 12V PWM :-)

Wow, but then I wonder if the rails are really the issue in my case. Perhaps the rise time is the issue?

breezelunar, 2025-04-02:
Hi Uhi22, just two quick questions, what is the incremental rate of increase of the EVSE present voltage during the preCharge phase? Also, at what phase in the Req/Res communication do the actual positive and negative battery terminals make themselves available on the cable? Allowing current to flow?

Thanks

breezelunar, 2025-04-04:
Just an update, we are now able to charge our car at 10kW using an MPPT directly from our solar panels! 

Thanks for all the help, will keep you updated with further developments

uhi22, 2026-02-25:
No further open question here. Closing.