uhi22/pyPLC #38: Not be able to capture V2G messages

nyannguyen, 2025-01-09:
Hi, I have modified the TP Link PA 4010 followed the instruction. 

What I want is to capture all the communications between Vehicle and Charger.
However when I plug the CP and PE of the vehicle to the HomePlug and then connect the charging gun to vehicle, I can only capture via WireShark the SLAC messages and all the V2G messages are not displayed.

[image]

Do you have any suggestions?
Thank you

uhi22, 2025-01-09:
Hi, thanks for supporting me on Patreon.
Best place for discussing such topics would be the Openinverter Forum, the AR7420 thread. Because there we have multiple experts.
So just the short version here: I observed the same. Until now nobody found out, how to convince the modem to route all traffic in sniffer mode. We should see the SDP request, because this is a broadcast message. But everything afterwards is unicast and the modem does not forward it.
If you do not even see the SDP request, something is wrong with pyPLCs listen mode. PyPLC should extract the network key from the slac.match and configure the modem using setkey. Afterwards the modem should join the private network between the evse and the pev.

nyannguyen, 2025-01-09:
Hi, thanks for your response.

Actually in the photo you see that were the last messages I received. After that nothing..
My setup is just modify the Modem and then connect to my laptop, using wireshark to *listen* to the communication. Is there any other steps that I need to do like running the pyPLC in Listen mode? 

Thank you.

uhi22, 2025-01-10:
Ahh, ok, you just connected the modem, without pyPLC in ListenMode.
Then you only see the SLAC. Because the last step of the SLAC is the SLAC_MATCH, where the "secret" key is exchanged, and both modems are setting this key, to establish a "private" network. Other modems (like your sniffing modem) will not see any further traffic, because they do not use the key.
The step further would be to run pyPLC in ListenMode. In this mode, the pyPLC should extract the key from the SLAC_MATCH, and write the key into the sniffer modem using SetKey. Then the sniffer modem joins the "private" network. This should be visible on the modem, all three LEDs are on, if the network is joined.
Nevertheless this helps not much. Because in only allows to sniff the very beginning of the high level communication, the SDP request broadcast. The later traffic (TCP) was not visible in my experiments, because the modem knows that this traffic is not intended to be routed to the sniffer machine, I guess based on the MAC addresses.

uhi22, 2025-01-30:
Progress: latest commit enables the pyPLC in ListenMode to sniff the network key and set it to the modem, so that the modem joins the AVLN.
If the modem is fast enough, this allows sniffing of the SDP request. And some more things as discussed in the other issue.
Activation of the feature with smart_listener_configuration = 1