uhi22/pyPLC #39: Sniffing V2GTP packets ("Biggest Challenges" in readme)

M4GNV5, 2025-01-17:
Hey,

we have been partially successfull in convincing the modem to forward packets which were ment for the car to us.
The trick was to send a fake packet which looks like its originating from the car to some random/broadcast address.
This makes the modem think the device with the cars mac address is in fact connected to its ethernet port and thus forwards packets ment for the car to its ethernet side.

Sadly this approach seems to also affect the cars modem and thus it does not forward packets ment for the car to the cars secc anymore i.e. breaking charging communication / causing a lot of retransmissions.

One solution could be to somehow (physically) prevent the modem from actually sending something, but allowing to receive. Somewhat like a diode, but a classical diode probably does not work with the HF powerline signal.
We did some tests with different PIB options to prevent it from sending (i.e. changing the TxGain), but either we did something wrong, or none of them seem to have any effect. I also tried [flooding the MAC cache](https://en.wikipedia.org/wiki/MAC_flooding) but also without any luck.

Let me know if you have any idea on how to make this work.

You can find our script for spoofing the routing [here](https://github.com/securityinmobility/v2g-sniffing/blob/main/qca-spoofing-from-pev.py). And for flooding the cache [here](https://github.com/securityinmobility/v2g-sniffing/blob/main/qca-mac-cache-flooding.py)

M4GNV5, 2025-01-17:
Just seeing the other issue now. This is related to #38

uhi22, 2025-01-17:
Hi, thanks for continuing the investigations. I also tried mac flooding without success, so we have the same result.
Blocking any transmission physically is easy. The modem chip has separate pins for the receive and the transmit path. So cutting the transmit path is possible by desoldering two capacitors or literally cutting two copper tracks. But, this also prevents the management traffic between the modems, which may hurt or help, it is for sure worth a try.

uhi22, 2025-01-17:
BTW Which type of modem did you use?

M4GNV5, 2025-01-17:
We have an [EVchargeSE](https://chargebyte.com/controllers-and-modules/evse-controllers/evacharge-se) which can do PWM/resistor communication behaving as either car or evse, but comes with a crappy old embedded linux soc and we have two [red beet eval boards](https://www.codico.com/en/red-beet-eval-board-e-2-0-evse) (one evse, one pev).

M4GNV5, 2025-01-17:
sorry for the closing. I misclicked. Feel free to re-open (I dont have permissions).

We have an [EVchargeSE](https://chargebyte.com/controllers-and-modules/evse-controllers/evacharge-se) which can do PWM/resistor communication behaving as either car or evse, but comes with a crappy old embedded linux soc and we have two [red beet eval boards](https://www.codico.com/en/red-beet-eval-board-e-2-0-evse) (one evse, one pev).

I used the redbeet for the tests. it even comes with simple dip switches for disconnecting RX/TX (mainly used for switching between PLC on AC and PLC on PWM). We tried disconnecting TX yesterday, but the communication was still the same.
The communication even works with all dips in off position but with a lot higher attenuation.

[image]

I assume this is because of the plc communication behaving like an antenna transmitting the signal "through" the closed DIP switches. The [brokenwire](https://www.brokenwire.fail/) stuff is also somewhat based on this.

Maybe grounding the TX pin works. We will try this next week (Working on this as part of my employment at Technische Hochschule Ingolstadt)

M4GNV5, 2025-01-20:
I did some further testing but the wireless transmission made it hard to get it working reliably.
Disconnecting TX seems to break the whole communication (receiving and sending).
I also tried grounding TX, but I think the ground wiring between TXN and TXP behaved like an antenna making it actually more reliable than simply disconnecting TX:

[image]

I am starting to think the modems exchange some form of message delivery information with each other. For some reason I sometimes get a "GetBridgeInformationsConfirmation" homeplug AV packet.
I am somewhat out of ideas. I will upload the pcap files to our logs repo in case anyone is interested.

You are welcome to drop by for a coffee and/or lab session any time. Maybe together we can get this to work. I assume you are working during the day, so we would have to move this to the evening?
Feel free to contact me at jakob.loew@thi.de for exchanging phone numbers or something 😊

uhi22, 2025-01-21:
I tested the scenarios "spoofing PEV MAC" and "cutting the TX path". Details collected here: https://github.com/uhi22/pyPLC/blob/master/doc/sniffing_experiments.md
Summary:
* Spoofing the MAC leads to be best results I've ever seen until now. It reveals sporadically some V2GTP frames, but the procedure stops the charging session sooner or later, as already describe above.
* Cutting the TX path on a AR7420 modem works. Listening to the SLAC is still possible, but joining the network does not work, because most likely for joining there is exchange of some management packets necessary.

Maybe studying the homeplugAV specification gives some hints. Not sure whether this is public.

M4GNV5, 2025-01-23:
I do not think we need to manually craft a homeplug packet in this case: The [int6krule](https://man.cx/int6krule(1)) tool included in open-plc-utils seems to be a cli tool to configure these classifier rules.
Unfortunately the actions which can be configured do not seem to include anything related to forwarding packets which are not ment for us.

Maybe we can use Drop and/or TagTX for preventing the other modems from receiving the packets and thus not breaking communication when doing MAC spoofing.

uhi22, 2025-01-23:
Maybe the -r to read the rules gives a hint. From how I understand the rules now, I would assume our listener device automatically creates a rule "drop everything which has destination A or destination B". If such rules exist, the task is just to remove this rule, so that the messages are not dropped anymore.

uhi22, 2025-01-26:
Added some Investigation results regarding classifier rules here: https://github.com/uhi22/pyPLC/blob/master/doc/sniffing_experiments.md#q-does-the-sniffer-modem-contain-automatic-rules-to-filter-out-the-not-wanted-traffic

M4GNV5, 2025-01-27:
I did some tests myself. It seems like you can configure the modems "firewall" rules even from outside. I can control these rules of our Tesla through the CCS port even if no PWM generation is running:

[image]
Side note: In our case the tesla evcc has MAC `98:ed:5c:93:0d:65`, the powerline modem has `98:ed:5c:93:0d:66`. I would assume other cars to also have the same MAC for the modem just incremented by one.

For our EvachargeSE the modem always denies readouts, but seems to accept setting/deleting rules. Also just like you described you can see they are saved, since removing them twice does not work.
[image]

Unfortunately according to my tests all of this does not help 😒 
I tried setting a DropRX rule on the tesla to prevent my mac spoofing packets from altering the PLC<->ETH routing table and I tried setting DropTX rules on my sniffer PLC modem for preventing it from sending the mac spoofing packets out to PLC.

It looks like the filter rules are applied one layer "above" the routing (think of an ISO/OSI stack, which the rule checking being one layer above the routing). This means for the transmitting side they are applied before routing and for receiving they are applied after routing.
This means when trying the DropRX on the Tesla modem the routing is still affected since it is updated before packets are dropped => The Tesla Modem still thinks the Tesla MAC is on the PLC side and stops forwarding packets to the Ethernet side.
When doing DropTX on the sniffing modem the routing is not affected since it is dropped before the routing happens => The Sniffing Model still thinks Tesla MAC is on the PLC side and never forwards those packets to us.

M4GNV5, 2025-01-27:
Heureka 🎉 

Got this working with [this script](https://github.com/securityinmobility/v2g-sniffing/blob/main/sdp-mitm.py). You can find the PCAP logs [here](https://github.com/securityinmobility/vehiclelogs/blob/main/vehicle2grid/2025-01-27-first-successfull-mitm-with-power.pcapng)
Or at least somewhat, since it is now based on MITM rather than Sniffing, but I got it working as described in the last edit above.

I let the PEV do SLAC, then as soon as I receive the SLAC_MATCH_CNF I configure a filter on the EVSE to drop all packets which are not sent by me.
I then spoof IPv6 Neighbor Solicitations and forward all other Ether packets from EVSE to PEV and from PEV to EVSE.

In theory this would also be possible without the DropRX filter, but then it relies on transmission speed and might still cause problems. Thanks to the DropRX filter we can make sure there is no communication possible between PEV and EVSE.

In the future someone (@eder-lukas 👀) could/will integrate this with the automatic AVLN joining, such that upon receival of a SLAC_MATCH_CNF:
- we configure DropRX on the EVSE
- we join the AVLN using SET_KEY on our own modem
- we forward packets between PEV and EVSE

uhi22, 2025-01-27:
Sounds like permanent rules could be used to prevent a car charging on a certain charger, or vice versa. Ooops.

M4GNV5, 2025-01-27:
I would assume a generic drop all rule could also prevent a car from charging at *any* charger, but sadly Tesla does not award bug bounties for denial of service :(

I did not try setting values without joining an AVLN. For our custom charging station we always use the same NID/NMK and thus joining the AVLN once is enough for all tests. I have seen some commercial ones do this as well (superchargers if I recall correctly). Others might change their NID/NMK for each charging session though.

I would not be sure if a permanent rule is even needed. From my understanding the rule is stored in pib SDRAM, which should survive the next charging session as long as the modem is not powered off completly. 
The bigger problem is to find out the modem MAC address of the charging station. For our Tesla we found the modem by luck: When trying the mac spoofing it sent a bridge information packet for some reason. Not sure if there is a reliable way to list modem MAC addresses though.

uhi22, 2025-01-28:
Finding out the MAC addresses of the modems is easy. We send a broadcast "get software version", and get from each modem in the AVLN a response. Ok, then still the task is to find out, which of the three modems belongs to which participant.
Code: https://github.com/uhi22/pyPLC/blob/4914b2fdc8ed684b3d7779a9d9bc97f18617e48e/pyPlcHomeplug.py#L178
PCAP: https://github.com/uhi22/pyPLC/blob/master/results/2022-10-26_WP4_networkEstablishedButHiddenCommunication.pcapng

uhi22, 2025-01-30:
Great progress. And because everything good needs a name, I would call this approach the "Ingolstadt CCS Leak" or "Ingolstadt Leak" :-)
It is a good basis for further experiments, do you plan to publish further results somewhere, so that we could add a link here and close the specific issue?

bl0tzz, 2025-06-25:
Just commented on the forum but posting here also. I just found a few interesting looking documents from the very earliest versions of open-plc-utils:
https://github.com/qca/open-plc-utils/blob/5326657419b48a2cb9e3f826014537ca2f592076/docbook/nda.h.html
https://github.com/qca/open-plc-utils/blob/5326657419b48a2cb9e3f826014537ca2f592076/docbook/plcsnif.c.html
https://github.com/qca/open-plc-utils/blob/5326657419b48a2cb9e3f826014537ca2f592076/docbook/plcsnif.7.html
https://github.com/qca/open-plc-utils/blob/5326657419b48a2cb9e3f826014537ca2f592076/docbook/Sniffer.c.html

Definitely worth a read. I haven't built or tested this early version with the missing files recreated from these .html, but looks promising.

uhi22, 2026-09-17:
Breaktrough: Transparency of the QCA7005 is reached with dedicated firmware. https://github.com/uhi22/ccs32dave

M4GNV5, 2026-09-17:
Very nice progress! And also cool video. I think I recognize that charger near Ingolstadt Nord 👀 

Any details on where you "found" the pingpong firmware? Or did you manually modify a stock firmware?

While the 2023 QCA firmware on the alpitronic might sound "new", it is actually old / too old. The colleagues from Oxford/Manchester university as well as some independent research labs have discovered multiple vulnerabilities with those modems. The latest even allows RCE: https://app.opencve.io/cve/CVE-2026-25293

uhi22, 2026-09-17:
Yes, this is indeed at the AGIP Römerstraße :-)

Not sure about how much detail is suitable to be published without getting in troubles. So only the birds-view here:
The stock firmware is a good basis, it contains everything we need, nearly, only the following four features are missing, so extended it:
- SLAC traffic routing depends on the configured role (PEV or EVSE). Feature added: Route both sides.
- NMK needs to be taken from slac_match and be applied in the same way as the cm_set_key does it. Feature added.
- IP traffic routing happens only for own or broadcast destination TEI, and the "own" is just one setting, not possible to set PEV and EVSE TEI at the same time. Feature added: Look at the traffic, and decide which TEI will be most likely the next of interest. This needs care, because the IPv6 traffic is basically ping-pong on V2G layer, but on TCP there are also pure ACK, which are out of interest, and are not covered by the "prepare the direction for the next V2G" approach.
- The read-memory and write-memory services are disabled and address-limited per default. Feature added: valuable developer services activation
All these patches required address space in the flash image, so we needed to hunt for "unused" code, which can be overwritten without side effects. Some bigger service handlers which are never used in the test setup provided this room.
An interesting point is the flash image creation. The flash image contains multiple sections, like secondary loader, parameter set, compressed main application, partly doubled-up, with CRCs, linked blocks. Patching the main application means: decompress the firmware (algorithm is a standard one), add the patch, compress, fit back to the container. The patching changes the size of the compressed images, so it does not fit into the orginal container size, and the loader even ignores too-small compressed images. So a strategy to keep the exact compressed size needed to be applied: Tune some unused code area by writing low-entropy and high-entropy data in a way, that the compressed image has the exact size of the original one.

This does not really mean "manually patched". ClaudeCode did this for us, disassembled and explained the interesting places, controlled the debugger for live investigation, controlled the other two modems on the bench, the power supply to get out of stuck state due to debugging, proposed assember patches, translated the asm into binary via python keystone, partly simulated it before applying, using unicorn, and then created the compressed app and full image, including CRCs. And flashed the image via flashrom. Full autonomous loop.

I started to try such a patching approach manually some months ago, but you get crazy maintaining already some-ten bytes, controlling the debugger, recover and re-write after a crash. So automation is the key, and AI driven automation solved this in a few days, with human just given a few directions, asking helpful questions, and - mainly - care for the bench that each corner-case is considered in the setup and AI has a good view of what happens and is able to recover from each situation autonomously.

uhi22, 2026-10-09:
The above success story was regarding the QCA7005 transparency.
Now also the AR7420 is transparent: https://github.com/uhi22/pyPLC#2026-10-09-ar7420-full-transparency-sniffer